open62541/open62541 #3951: The ServiceResult has the StatusCode BadInternalError

Yashvanthb13, 2020-10-08:
I have Generated the certs using create_self-signed.py in tools/cets, i have not change any configuration in the localhost.cnf file
i add the certs file location in code 
 

	#include <open62541/client_config_default.h>
	#include <open62541/client_highlevel.h>
	#include <open62541/client_subscriptions.h>
	#include <open62541/plugin/log_stdout.h>

	#include "common.h"

	#define NONE                         (char *)"http://opcfoundation.org/UA/SecurityPolicy#None"
	#define Basic256Sha256               (char *) "http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256"
	#define Basic256                     (char *)"http://opcfoundation.org/UA/SecurityPolicy#Basic256"
	#define Basic128Rsa15                (char *)"http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15" //(no longer considered as secure)None http

	int main(int argc, char *argv[])
	{
		UA_StatusCode retval = 0;

		char *serverurl = "opc.tcp://localhost:4840", *username = "test", *password = "test";
		
	#ifdef UA_ENABLE_ENCRYPTION
		char *certfile = "..\\certs\\server_cert.der";
		char *keyfile = "..\\certs\\server_key.der";
	#endif

		/* Create the server and set its config */
		UA_Client *client;
		
		client  = UA_Client_new();
		
		UA_ClientConfig *cc = UA_Client_getConfig(client);

		 
		UA_String securityPolicyUri = UA_STRING(Basic256Sha256);

		UA_MessageSecurityMode securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT; /* allow everything */

	#ifdef UA_ENABLE_ENCRYPTION
		if(certfile)
		{
			UA_ByteString certificate = loadFile(certfile);
			UA_ByteString privateKey  = loadFile(keyfile);
			cc->securityMode = securityMode;
			cc->securityPolicyUri = securityPolicyUri;

			retval =  UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey, NULL, 0, NULL, 0);
		 
			if(retval == UA_STATUSCODE_GOOD) 
			{
				printf("Client has been configured in Default Mode with security\r\n");
			} 
			else 
			{
				printf("Failed to confgure Client in Default Mode with security\r\n");

			}

			UA_ByteString_clear(&certificate);
			UA_ByteString_clear(&privateKey);
		}
		else 
		{
			retval = UA_ClientConfig_setDefault(cc);
			if(retval == UA_STATUSCODE_GOOD) 
			{
				printf("Client configured in Default setting\r\n");

			} 
		}
	#else
		UA_ClientConfig_setDefault(cc);
	#endif

		/* Connect to the server */
		//retval = UA_STATUSCODE_GOOD;
		
		if(retval == UA_STATUSCODE_GOOD)
		{
			retval = UA_Client_connectUsername(client, serverurl, username, password);
			if(retval != UA_STATUSCODE_GOOD) 
			{            
				printf("Unable to connect with username and password retlvalue\t:\t%02X \r\n",retval);
			} 
			else 
			{
			  printf("Connection established with server using Username and password\r\nretval \t:\t%02X\r\n", retval);
			}
		} 
		else 
		{
			retval = UA_Client_connect(client, serverurl);
			printf("Connection Established with server without Username and "
				   "password\r\nretval \t:\t%02X\r\n",
				   retval);
		}

		if(retval == UA_STATUSCODE_GOOD) {

			UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Connected!");

			/* Read the server-time */
			UA_Variant value;
			UA_Variant_init(&value);
			UA_Client_readValueAttribute(
				client, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME),
				&value);
			if(UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
				UA_DateTimeStruct dts = UA_DateTime_toStruct(*(UA_DateTime *)value.data);
				UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
							"The server date is: %02u-%02u-%04u %02u:%02u:%02u.%03u", dts.day,
							dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
			}
			UA_Variant_clear(&value);
		}
		/* Clean up */
		UA_Client_disconnect(client);
		UA_Client_delete(client);
		return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
	}

LOG when connecting with Pro sys server 

	2020-10-08 19:01:37.718 (UTC+0550)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
	Client has been configured in Default Mode with security
	[2020-10-08 19:01:37.724 (UTC+0550)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
	[2020-10-08 19:01:37.748 (UTC+0550)] info/channel       Connection 388 | SecureChannel 20 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
	[2020-10-08 19:01:37.749 (UTC+0550)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
	[2020-10-08 19:01:37.761 (UTC+0550)] info/client        Rejecting endpoint 0: security mode doesn't match
	[2020-10-08 19:01:37.762 (UTC+0550)] info/client        Rejecting endpoint 1: security policy doesn't match
	[2020-10-08 19:01:37.762 (UTC+0550)] info/client        Rejecting endpoint 2: security mode doesn't match
	[2020-10-08 19:01:37.762 (UTC+0550)] info/client        Rejecting endpoint 3: security mode doesn't match
	[2020-10-08 19:01:37.762 (UTC+0550)] info/client        Rejecting endpoint 4: security policy doesn't match
	[2020-10-08 19:01:37.762 (UTC+0550)] info/client        Rejecting endpoint 5: security policy doesn't match
	[2020-10-08 19:01:37.762 (UTC+0550)] info/client        Rejecting endpoint 6: security mode doesn't match
	[2020-10-08 19:01:37.763 (UTC+0550)] info/client        Rejecting endpoint 7: security policy doesn't match
	[2020-10-08 19:01:37.763 (UTC+0550)] info/client        Rejecting UserTokenPolicy 0 (anonymous) in endpoint 8: configuration doesn't match
	[2020-10-08 19:01:37.763 (UTC+0550)] info/client        Rejecting UserTokenPolicy 1 (certificate) in endpoint 8: configuration doesn't match
	[2020-10-08 19:01:37.763 (UTC+0550)] info/client        Rejecting UserTokenPolicy 2 (certificate) in endpoint 8: configuration doesn't match
	[2020-10-08 19:01:37.763 (UTC+0550)] info/client        Selected Endpoint opc.tcp://DESKTOP-C5TCGL0:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
	[2020-10-08 19:01:37.763 (UTC+0550)] info/client        Selected UserTokenPolicy username_basic128 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
	[2020-10-08 19:01:37.764 (UTC+0550)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
	[2020-10-08 19:01:38.116 (UTC+0550)] info/channel       Connection 388 | SecureChannel 21 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 600.00s
	[2020-10-08 19:01:38.116 (UTC+0550)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
	[2020-10-08 19:01:38.136 (UTC+0550)] info/client        Received a ServiceFault response
	[2020-10-08 19:01:38.136 (UTC+0550)] info/client        The ServiceResult has the StatusCode BadInternalError
	[2020-10-08 19:01:38.136 (UTC+0550)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: BadInternalError
	Unable to connect with username and password retlvalue  :       80020000
	[2020-10-08 19:01:38.136 (UTC+0550)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError

in pro sys server Application i trusted the certificated of open642541 
[image]
[image]

can anyone tell what went wrong?

Yashvanthb13, 2020-10-14:
Can one help on this i got stuck in this stage.

jpfr, 2020-10-20:
Try to get more detailed log output from prosys. Difficult to say otherwise.

Yashvanthb13, 2020-10-20:
[image]
[image]

This are logs of prosys opc ua server.

the user name and password is "test"

I generated the certs using the create_self-signed.py file and included the path. 
I have not changed anything in the localhost.cnf

jpfr, 2020-10-20:
Little to see.
Can you increase the log verbosity in both sides?

Yashvanthb13, 2020-10-20:
[image]

This are the logs of today when i try to connect to the prosys server with certs and username and password

SplitGemini, 2022-05-13:
Any progress, with [issues/5036](https://github.com/open62541/open62541/issues/5036)

andreasebner, 2025-07-30:
CLOSE: The URIs in the application and certificated have to match. There's been a big overhaul in the crypto system, so I'll close this issue.

closed 2025-07-30 by andreasebner