open62541/open62541 #3953: Example to make Pubsub on both side

Waner98, 2020-10-09:
Hi every one,

Do you know if there is an example which give you both client and server to make some PubSub relation? 
Moreover there is something that I do not understand in the documentation. What is the difference between the Publishment Service / Subscription Service, and the publish/subscribe in the "internal" field? Is the last one higher level? How should I process to follow some steps in order to make my Ethernet PubSub between  Client and a Server? 

```bash
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2021-04-22:
Inside the "examples/pubsub" folder, there is publish/subscribe example (tutorial_pubsub_publish, tutorial_pubsub_subscribe)
The PubSub subscribe example demonstrates the simplest way to receive information over two transport layers such as UDP and Ethernet, that are published by tutorial_pubsub_publish example.

Note:
The PubSub Subscriber API is currently not finished. This Tutorial will be continuously extended during the next PubSub batches.

NoelGraf, 2021-04-29:
I would close the issue for now. Please reopen if required!

closed 2021-04-29 by NoelGraf