open62541/open62541 #3956: addObjectNode adds an "empty" object

JensMertelmeyer, 2020-10-14:
## Description
This is probably a misunderstanding on my side and not a technical issue. I am defining my own object type by using `UA_Server_addObjectTypeNode` and add a variable to this type.

When i create an "instance" of this new object type, it is there, but does not have the variable I added to its type definition.

I am following the tutorial [Working with Objects and Object Types](https://open62541.org/doc/1.1/tutorial_server_object.html#object-types-type-hierarchies-and-instantiation) and I see I can define a constructor for that type which I haven't. The tutorial sounds as if this optional.

## Background Information / Reproduction Steps
I am using the official precompiled release for 1.1.2 (Win64).

I have attached the full source code (70 lines), but here is the gist of it:

I add my object type
```cpp
UA_Server_addObjectTypeNode(
	server,
	UA_NODEID_NULL,
	UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
	UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
	UA_QUALIFIEDNAME(1, "t_MyStruct"),
	typeAttributes,
	nullptr,
	&NodeIds::_Types::t_MyStruct		
);
```
and then add a variable node to that type:
```cpp
UA_Server_addVariableNode(
	server,
	UA_NODEID_NULL,
	NodeIds::_Types::t_MyStruct,
	UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
	UA_QUALIFIEDNAME(1, "TimeStamp"),
	UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
	timestampAttributes,
	nullptr,
	nullptr
);
```

Finally, I add an object node:
```cpp
UA_Server_addObjectNode(
	server,
	UA_NODEID_NULL,
	UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
	UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
	UA_QUALIFIEDNAME(1, "MyStruct"),
	NodeIds::_Types::t_MyStruct,
	objectAttributes,
	nullptr,
	&NodeIds::_Objects::MyStruct		
);
```

All methods return `UA_STATUSCODE_GOOD`

Browser correctly show the defined type which inherits from `BaseObjectType`
[image]
Howev,er, the "instance" is empty and lacks the "Timestamp" variable:
[image]

Here is the full source code:
https://gist.github.com/JensMertelmeyer/0f6c4af370e0ce4e9faed36d80471584

## Checklist
Name|Value
---|---
version | v1.1.2 (8ad8496)
Other OPC UA SDKs used | Unified Automation uaExpert, Prosys OPC UA Browser
Operating system|Windows, 64 bit
Compiler|C++, mingw-w64
Logs attached | Not necessary
Wireshark network dump| Not necessary
Self-contained code example attached | [Here](https://gist.github.com/JensMertelmeyer/0f6c4af370e0ce4e9faed36d80471584)
Critical issue | No

Many thanks in advance.

ccvca, 2020-10-16:
It looks like you the moddeling rule mandatory is missing:
https://reference.opcfoundation.org/v104/Core/docs/Part3/6.4.4/#6.4.4.4

JensMertelmeyer, 2020-10-16:
That's it, thank you very much 🥳

I needed to add a
```cpp
	// Make timestamp mandatory
	statusCode = UA_Server_addReference(
		server, 
		timestampNode,
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
		UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY),
		/* isForward: */true
	);
```
and my life was not in ruins anymore.

closed 2020-10-16 by JensMertelmeyer