open62541/open62541 #3962: Server; pass full UA_ActivateSessionRequest reference to userland activateSession callback

juangburgos, 2020-10-19:
## Extension to access control

This is just a simple change request to the server API; to pass the full `UA_ActivateSessionRequest` structure to [the userland access control callback for `activateSession`](https://github.com/open62541/open62541/blob/ce890e9d332450ba303eb6f59e623d8050ec2991/src/server/ua_services_session.c#L681). 

Specifically, pass the full `const UA_ActivateSessionRequest *request` instead of only `const UA_ExtensionObject * userIdentityToken` as the 6th argument of the callback. 

This way users could implement other types of authentication, for example x509 certs, which currently cannot be done without access to the `userTokenSignature` field of the request.

Thanks,

jpfr, 2020-10-20:
We currently do all cryptographic verification in the main library and not in the accesscontrol plugin.

Also the password is decrypted before passing it to the plugin.

Does the userTokenSignature hold authentication credentials or is it just a signature to be verified?

Answering from mobile. So don’t have the standard at hand atm.

juangburgos, 2020-10-20:
TLDR; As I understand it, if using x509 for authentication (`X509IdentityToken`), during *CreateSession* the server sends a the `serverNonce` on the response. The client should sign it with its private key and send it during *ActivateSession* in the `userTokenSignature` of the request.

Yes, I saw from the open62541 code that it decrypts the passwrod, but only if the token type is `UserNameIdentityToken`, this does not apply for tokens of type `X509IdentityToken`.

I personally think the easiest is to pass the full reuqest to the plugin, because the spec also defines `IssuedIdentityToken` with in principle should allow for custom auth services. So the user of the library who would like to use some custom services might need all the request data.
 
Below some relevant extract from the spec;

## 5.6.2 CreateSession 

The Session created with this Service shall not be used until the Client calls the ActivateSession Service and provides its SoftwareCertificates and proves possession of its Application Instance Certificate and any user identity token that it provided.

### Response Parameters

* **serverNonce** (ByteString)

A random number that should never be used in any other request.
This number shall have a minimum length of 32 bytes.
The Client shall use this value to prove possession of its Application Instance Certificate in the ActivateSession request.
This value may also be used to prove possession of the userIdentityToken it specified in the ActivateSession request.

## 5.6.3 ActivateSession 

When a Client provides a user identity then it shall provide proof that it is authorized to use that user identity.
The exact mechanism used to provide this proof depends on the type of the `UserIdentityToken`. 
If the token is a `UserNameIdentityToken` then the proof is the password that is included in the token. 
If the token is an `X509IdentityToken` then the proof is a **signature generated with private key** associated with the Certificate. 
The data to sign is created by appending the last **serverNonce** to the serverCertificate specified in the `CreateSession` response. 
If a token includes a secret then it should be encrypted using the public key from the serverCertificate.

### Request Parameters

* **userIdentityToken** (Extensible Parameter UserIdentityToken)

The credentials of the user associated with the Client application. 
The Server uses these credentials to determine whether the Client should be allowed to activate a Session and what resources the Client has access to during this Session.
The `UserIdentityToken` is an extensible parameter type defined in 7.36.
The `EndpointDescription` specifies what `UserIdentityTokens` the Server shall accept.
Null or empty user token shall always be interpreted as anonymous.

* **userTokenSignature** (SignatureData)

If the Client specified a user identity token that supports digital signatures, then it shall create a signature and pass it as this parameter. Otherwise the parameter is null.
The `SignatureAlgorithm` depends on the identity token type.
The `SignatureData` type is defined in 7.32.

The Server validates the signatures provided with the request and then validates the new user identity. 
If no errors occur the Server replaces the user identity for the Session.

jpfr, 2020-10-20:
Since the signature requires access to the ServerNonce, we have to do the check internally. The plugin has no access to the nonce via the public API.

Feel free to add the test for the Signature in a PR. For now, let’s add x509 auth support without changing the API.

Once we have a second case that requires the signature check, we can think about generalizing it.

juangburgos, 2020-10-20:
Indeed the *nonce* is not in the public API, but at least there is a way to *hack* it because it gets stored temporarily in the `UA_Session` instance (`->serverNonce`). For example, In [QUaServer](https://github.com/juangburgos/QUaServer/blob/master/src/wrapper/quaserver_anex.h) I use a lot of the internal open62541 API with out the need of modifying the open62541 source code, simply by adding an extra header exporting the stuff that I need. 

I have used that technique for example, to hack the event handling in order to integrate my wrapper deeper into open62541 without having to maintain a fork.

In my case, having the full activate session request will suffice, because I have used the hack to expose the full session object to my wrapper, thus I can access the session nonce. But in the case of the session services, where the authentication is done, they are too deep into the open62541 library, so that hack won't cut it.

jpfr, 2020-10-20:
So?

juangburgos, 2020-10-20:
It shall be enough to pass the full request to the plugin, to support not only x509 but also custom auth providers. And if at some point in the future that would eb acceptable, I would go a little further and expose the full `UA_Session` to the API, so in all use land callback instead of having a lot of `&session->sessionId, session->sessionHandle, ...` arguments, just have a `&session` argument.

Would open up many more possibilities for users. IMHO.

jpfr, 2020-10-20:
The whole idea of having public and private API separate is to have a stable public API and to hide internals.

It seems there is no real disadvantage in testing the signature internally. On the plus side, it separates the auth mechanism from crypto concerns.

So let’s avoid breaking the API for all the auth. plugins that are already out there.

juangburgos, 2020-10-20:
So, if I understand correctly; I implement the signature testing in `Service_ActivateSession` (using cypto stuff), then we leave it to the user to implement that the user's cert is trusted in the access control plugin?

I am no c developer, but I can give it a try.

MichaelKlin, 2021-03-25:
@juangburgos Hi, where do you took spec?

juangburgos, 2021-03-25:
> @juangburgos Hi, where do you took spec?

https://reference.opcfoundation.org/v104/

MichaelKlin, 2021-03-25:
@juangburgos  thank you

NoelGraf, 2021-04-22:
Can the issue be closed?

juangburgos, 2021-04-22:
Nope, merge request still open, x509 still unsupported

jpfr, 2021-09-24:
X509 is now merged.

closed 2021-09-24 by jpfr