open62541/open62541 #3964: Bug(NodeSet Compiler): Values in NodeSet (enums & optionset)

Fischer-Thomas, 2020-10-19:
## Description
Currently there is a bug when trying to compile a nodeset which includes nodes with values of custom OptionSets and enumerations. 
For enumerations the code generator tries to assign the string from the nodeset (for example **NanoSecond_0** instead of just using **0 as value** or the generated enumValue **UA_CUSTOMUNITENUM_NANOSECOND**).
And for the OptionSet it uses the wrong structure. (it doesn't use the members **Value** or **ValidBits** for assigning the values) 
Also the two values that should be assigned are generated as global variables inside the file with **the same name**.
I attached all relevant files i used to the issue.
 
[customNodeSet.zip](https://github.com/open62541/open62541/files/5401185/customNodeSet.zip)

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2021-04-22:
Does the problem still exist? Have you tested it with the current 1.2 version?

Fischer-Thomas, 2021-04-22:
The problem still exists. I tested it with the current master **and** 1.2 version.

Here are some screenshots:

The added object using the custom DataType.
[image]

Generated code: (OptionSet errorr)
[image]
[image]

Generated code: (Enumeration error)
[image]

jpfr, 2021-12-16:
Is this done with the latest changes to the nodeset compiler @andreasebner ?

andreasebner, 2021-12-16:
Should be solved by the merged PR's or the open #4715 @NoelGraf can you confirm?

andreasebner, 2025-07-30:
The mentioned PR is merged, so this should be fixed. Also, a lot of the companion specs are tested in our CI

closed 2025-07-30 by andreasebner