open62541/open62541 #3974: nodeset compiler generates errors with Machine Vision companion specification

DADev99, 2020-10-26:
## Description

I am trying to use the nodeset_compiler.py script the Machine Vision companion specification. When I run the script I get many warnings and errors.

## Background Information / Reproduction Steps
This is the command I use to run the script:
python .\nodeset_compiler\nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_MVMODEL --existing .\ua-nodeset\Schema\Opc.Ua.NodeSet2.xml --xml .\ua-nodeset\MachineVision\Opc.Ua.MachineVision.NodeSet2.xml mvmodel

This is the output in the console:
INFO:__main__:Preprocessing (existing) .\ua-nodeset\Schema\Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing .\ua-nodeset\MachineVision\Opc.Ua.MachineVision.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute IsOptional
WARNING:nodes:Unknown Field Attribute IsOptional
ERROR:datatypes:ns=1;i=6308: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6343: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6344: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6344: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6307: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6307: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6323: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6323: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6345: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6141: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6294: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6292: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6292: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6300: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6299: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6299: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6304: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6304: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6305: Expected XML element with tag StartTime but found EncodingMask instead
ERROR:datatypes:ns=1;i=6305: Expected XML element with tag EndTime but found StartTime instead
ERROR:datatypes:ns=1;i=6305: Expected XML element with tag AcquisitionDuration but found EndTime instead
ERROR:datatypes:ns=1;i=6143: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6143: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6295: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6203: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6194: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6194: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6201: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6201: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6199: Expected XML element with tag Id but found EncodingMask instead
ERROR:datatypes:ns=1;i=6199: Expected XML element with tag Description but found Id instead
ERROR:datatypes:ns=1;i=6204: Expected XML element with tag Id but found ExtensionObject instead
ERROR:datatypes:ns=1;i=6132: Expected XML element with tag HasTransferableDataOnFile but found EncodingMask instead
ERROR:datatypes:ns=1;i=6120: Expected XML element with tag HasTransferableDataOnFile but found EncodingMask instead
ERROR:datatypes:ns=1;i=6043: Expected XML element with tag HasTransferableDataOnFile but found EncodingMask instead
ERROR:datatypes:ns=1;i=6049: Expected XML element with tag State but found EncodingMask instead
ERROR:datatypes:ns=1;i=6049: Expected XML element with tag StateDescription but found State instead
INFO:__main__:Generating Code for Backend: open62541
WARNING:backend_open62541_nodes:Cannot encode dataTypeNode: ResultDataType for value of node ResultType ns=1;i=2002
WARNING:backend_open62541_nodes:Cannot encode dataTypeNode: ResultDataType for value of node <ResultVariable> ns=1;i=6168
INFO:__main__:NodeSet generation code successfully printed

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Operating system: Windows 10

GoetzGoerisch, 2020-12-01:
@xenonforlife can you help here?

simeonthefirst, 2020-12-01:
Same problem here.  

For the first error first line: 
The nodeset compiler expects the node 'Id' as defined in JobIdDataType, the datatype of ns=1;i=6308. But gets the ExtensionObject and cannot handle it. ( In the __parseXMLSingleValue(...) method in datatypes.py)
 
Any ideas or workarounds?

simeonthefirst, 2020-12-04:
Same Problem appears when compiling with cmake , incuding the NodeIds.csv and the .bsd  containing the decoded base64 String from the Opc.Ua.MachineVision.NodeSet2.xml.

But the XML Nodeset Compiler seems to output usable files. I can instantiate the VisionSystemType (automatically including the mandatory Nodes).

Further, i can instantiate the otional Node SystemState (i=6049 in the Errorlog.) of type SxstemStateDescriptionType. The state enum of the SystemState works without complication but when I try to write the description wich is a UA_String inside the SystemStateDescription struct, either from a Client or as below, i get an BadEncoding Error and a reconnection of the Client.

       UA_VariableAttributes variable_attr=UA_VariableAttributes_default;
        variable_attr.dataType = UA_NODEID_NUMERIC(2,UA_MYTESTNSID_SYSTEMSTATEDESCRIPTIONDATATYPE);       
        variable_attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

        UA_Variant value2;

        UA_SystemStateDescriptionDataType ssdt;

        UA_String desc =  UA_STRING("myDescriptionText");

        ssdt = (UA_SystemStateDescriptionDataType) {.state=UA_SYSTEMSTATEDATATYPE_NST_6, .stateDescription = &desc};

        UA_Variant_setScalar(&value2, &ssdt, &UA_TYPES_MYTESTNS[UA_TYPES_MYTESTNS_SYSTEMSTATEDESCRIPTIONDATATYPE]);
        variable_attr.value=value2;

        UA_NodeId SystemStateId;
        UA_Server_addVariableNode(server,
                                UA_NODEID_NUMERIC(nsIdx, 0),
                                visionSystemId,
                                UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                                UA_QUALIFIEDNAME(nsIdx, "SystemState"),
                                UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), 
                                variable_attr,
                                NULL,
                                &SystemStateId);

xenonforlife, 2020-12-07:
@GoetzGoerisch  sorry, neither I nor any of the implementers of OPC MV (I know) have used the NodeSet Compiler and/or open62541 up until now, therefore we have no experience whatsoever what the NodeSet Compiler is expecting. I don't really know how I can help here, with other SDKs implementers haven't reported any such problems until now.

Stasik0, 2021-05-26:
I'd like to bump this up as well here

GoetzGoerisch, 2021-05-26:
At least I can confirm that I have the same results on Linux like @simeonthefirst wrote.

I suggest you file a [MANTIS](https://github.com/OPCFoundation/UA-Nodeset#error-reporting) issue for the _NodeSet_ file, if @NoelGraf and the team confirm this is **NOT** an issue with the nodeset compiler.

teccheck, 2025-06-06:
I've tested the nodeset compiler with the Machine Vision spec and it works now