open62541/open62541 #3977: Make Writergroup non periodic

Waner98, 2020-10-27:
## Description

Hi everyone,

I am actually working on PubSub and I use the last functionalities of the library to do it (e.g : addPubSubConnection, addDataSetField etc...). 

It works fine for me but I am a little bit frustrated..

In addWriterGroup(UA_Server * server) function, we configure a periodic writerGroupConfig which send regularly information via our protocol (in my case UDP multicast) instead of having this system of Notification that we are waiting to for PubSub. 
The solution should be to remove this periodic part in order to replace it with another one asynchronous. For exemple, a callback or better, an function called by another Thread. 

I was very enthusiast with this idea but I have absolutely no idea of how I can implement this. I do not found enough documentation on the subject. 

Do someone have a solution ? 

PS : Thanks for the work of all the team on this opensource library. 

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.2-269-gef0c6873
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

pubsub-tsn-team-kalycito, 2020-10-27:
There is a custom callback option for publishing and subscribing which allows the user to publish/subscribe according to their needs. Have a look at pubsub_TSN_publisher.c and pubsub_TSN_loopback.c samples for your reference.

opcua-tsn-team-kalycito, 2020-10-27:
There is a custom callback option for publishing and subscribing which allows the user to publish/subscribe according to their needs. Have a look at pubsub_TSN_publisher.c and pubsub_TSN_loopback.c samples for your reference.

Waner98, 2020-10-27:
Thank you for your quick answer, I will explore it in the afternoon!

Waner98, 2020-10-27:
@opcua-tsn-team-kalycito  After having seeing the two examples, I am still not very confident in my way to understand what happen. 

If I well understand, the basic configuration for the WriterGroup (like in the tutorial) propose us to have by default, a period when your message will be send through the cloud depending of your protocol (in my case I use UDP). 

To solve my issue (I want to send my message only when I decide it), I have to code a callback which work by interruption and which is able to say to WriterGroup, "ok now I want you to send the msg even if it s not the period time". And you can do it by the "pubsubManagerCallback" member of WriterGroupConfig. Do I well understand? 

But the things that I don t get is the next one, what in your callback explicitly says that you want to send your message? I think that I don't really understand the mechanism of this pubsubManagerCallback, even if I pay attention to the examples that you mentionned in the last message. 

Thanks for your time

opcua-tsn-team-kalycito, 2020-10-30:
> To solve my issue (I want to send my message only when I decide it), I have to code a callback which work by interruption and which is able to say to WriterGroup, "ok now I want you to send the msg [...]

Yes, you are right. This pubsubManagerCallback allows you to have a user based publish/subscribe callback mechanism. From the examples, you can see the publisher thread publish packets to the network whenever [pubCallback()](https://github.com/open62541/open62541/blob/ef0c6873b03d372a468fcdb30990af7535da45a3/examples/pubsub_realtime/pubsub_TSN_publisher.c#L749) API has been called (here, it is called for every cycle time). You can use this method to publish the packets whenever required. (i.e., call pubcallback() when you need to publish).

Waner98, 2020-10-30:
@opcua-tsn-team-kalycito  Yes I just discovered this pubCallback() API yesterday and it solved perfectly my issued! Now I can use mechanism with other threads and mqueue msg which trigger the callback, and so the send of UDP packet. 
Thanks for your answers, I can close this issue :)

closed 2020-10-30 by Waner98