open62541/open62541 #3980: How to add/change url

MichaelKlin, 2020-10-28:
## Description
Hallo, i want to ask, how can I add URL like http://opcapi.place/example. 
## Background Information / Reproduction Steps
I find some variables for host and port, but nothing for URL. Thank you

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): last release 
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: windows

MichaelKlin, 2020-10-29:
@jpfr can you please give a reference or some help and can close issue

jpfr, 2020-10-31:
OPC UA does not use http.

Are you on the server or client side?

MichaelKlin, 2020-11-01:
@jpfr i am on server side

MichaelKlin, 2020-11-01:
@jpfr because in OPC UA gui client widget i see, that can connect with   opc.tcp:// and with   https://

MichaelKlin, 2020-11-01:
[image]

MichaelKlin, 2020-11-13:
Hallo @jpfr , last question, there is no opportunity to use http? Bcause I do not know, there is field for Url and http. Sorry for my English

EugenM-sie, 2020-11-15:
Hallo @Pro can you maybe please help and i close this issue

MichaelKlin, 2020-11-15:
Hallo @Pro can you maybe please help and i close this issue

closed 2021-04-21 by EvgKlin