open62541/open62541 #3982: Issue building the server_ls example

Andri-e, 2020-10-29:
## Description
Hello, trying to make a discovery server (LDS) so I can register my simple server test on there and connect my client to other servers then just the default one but I ran into some problems running the server_lds.c example.

## Background Information / Reproduction Steps
Using a Raspberry pi 3b+ for this project and I installed Open62541 according to the descriptions on 

https://open62541.org/doc/current/building.html#build-options

at first I installed it with without Discovery enabled but then I went back later and enabled it using both ccmake .. and 

cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_DISCOVERY=ON ..

to enable the discovery, afterwards I re-did make and then sudo make while trying to figure out why it was not working. 

Then I copied the server_lds.c file over to my project folder on the dekstop along with the open62541.c and open62541.h file from the build folder, there I tried to build the server_lds.c file but got this error 

[image]

when I replace 

#include <open62541/server.h>
#include <open62541/server_config_default.h>

with 

#include <open62541.h>

I get (This worked with some other examples I had been playing with)

[image]

and I get the same issue when I try to compile it directly in the examples folder. 

Any idea what I might be doing wrong? 

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_DISCOVERY=ON ..
```

## Checklist

Please provide the following information:

 - [https://github.com/open62541/open62541.git ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ RaspberrY Pi Os 32 bit] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-10-31:
Is the lds example for the same release of the SDK?

Andri-e, 2020-11-01:
Yes I believe so, its the example folder that was cloned along with the SDK

closed 2020-11-11 by Andri-e