open62541/open62541 #3984: GetEndpointRequest failed with error code BadShutdown

JosephKhattar, 2020-10-30:
## Description

Hello i am a student , i am trying to establish a OPCUA between a safety chip with Uclinux operating system and my PC with Centos operating system.

## Background Information / Reproduction Steps

I cloned the library and followed the documentation and using the codes from the tutorials. The server run fine on the chip but the client is crashing and closing connection. The error is "GetEndpointRequest failed with error code BadShutdown" but when i debbuged i found that the value of the "UA_STATUSCODE_GOOD" is 0x800A0000 related to "UA_STATUSCODE_BADTIMEOUT",
I am having troubles to fix this issue.

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=MinSizeRel -DUA_NAMESPACE_ZERO=MINIMAL -DUA_LOGLEVEL=100 -DUA_ENABLE_AMALGAMATION=ON .. ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): https://github.com/open62541/open62541.git
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Server : Uclinux ,, Client :Centos
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-10-31:
The value of a good statuscode is zero.
Can you show some log output of client and server?

JosephKhattar, 2020-11-06:
Thank you for your reply , the problem was solved,  turned out there was no mistakes in any of the codes or configuration , the problem was in my Safety chip MTU (maximum transmission unit), i increased the data transmission per frame , and it worked perfectly. Thanks for your help.

closed 2020-11-06 by JosephKhattar