open62541/open62541 #3985: Warnings as errors build failure

je1h, 2020-10-30:
## Description

I'm trying to build open62541 inside docker container with alpine:3.11 base image.

Locally, on Ubuntu 20 host OS it builds fine with Debug and Release build modes both.

In docker container it builds fine in Debug build mode, while in Release build mode it fails with following errors:

```bash
In file included from /app/build/_deps/open62541-build/open62541.h:384,
                 from /app/build/_deps/open62541-build/open62541.c:28:
/app/build/_deps/open62541-build/open62541.c: In function 'UA_PubSubChannelUDPMC_receive':
/app/build/_deps/open62541-build/open62541.h:417:36: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
  417 | # define UA_fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
/app/build/_deps/open62541-build/open62541.c:65488:9: note: in expansion of macro 'UA_fd_set'
65488 |         UA_fd_set(channel->sockfd, &fdset);
      |         ^~~~~~~~~
/app/build/_deps/open62541-build/open62541.c: In function 'connection_recv':
/app/build/_deps/open62541-build/open62541.h:417:36: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
  417 | # define UA_fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
/app/build/_deps/open62541-build/open62541.c:68243:5: note: in expansion of macro 'UA_fd_set'
68243 |     UA_fd_set(connection->sockfd, &fdset);
      |     ^~~~~~~~~
/app/build/_deps/open62541-build/open62541.c: In function 'setFDSet':
/app/build/_deps/open62541-build/open62541.h:417:36: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
  417 | # define UA_fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
/app/build/_deps/open62541-build/open62541.c:68568:9: note: in expansion of macro 'UA_fd_set'
68568 |         UA_fd_set(layer->serverSockets[i], fdset);
      |         ^~~~~~~~~
/app/build/_deps/open62541-build/open62541.h:417:36: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
  417 | # define UA_fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
/app/build/_deps/open62541-build/open62541.c:68575:9: note: in expansion of macro 'UA_fd_set'
68575 |         UA_fd_set(e->connection.sockfd, fdset);
      |         ^~~~~~~~~
/app/build/_deps/open62541-build/open62541.c: In function 'UA_ClientConnectionTCP_poll':
/app/build/_deps/open62541-build/open62541.h:417:36: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
  417 | # define UA_fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
/app/build/_deps/open62541-build/open62541.c:68875:9: note: in expansion of macro 'UA_fd_set'
68875 |         UA_fd_set(clientsockfd, &fdset);
      |         ^~~~~~~~~
/app/build/_deps/open62541-build/open62541.c: In function 'UA_ClientConnectionTCP':
/app/build/_deps/open62541-build/open62541.h:417:36: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
  417 | # define UA_fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
/app/build/_deps/open62541-build/open62541.c:69127:13: note: in expansion of macro 'UA_fd_set'
69127 |             UA_fd_set(clientsockfd, &fdset);
      |             ^~~~~~~~~
/app/build/_deps/open62541-build/open62541.c: At top level:
cc1: error: unrecognized command line option '-Wno-static-in-inline' [-Werror]
cc1: all warnings being treated as errors
make[3]: *** [_deps/open62541-build/CMakeFiles/open62541-object.dir/build.make:278: _deps/open62541-build/CMakeFiles/open62541-object.dir/open62541.c.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:1062: _deps/open62541-build/CMakeFiles/open62541-object.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:131: CMakeFiles/gateway.dir/rule] Error 2
make: *** [Makefile:199: gateway] Error 2
```

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_LOGLEVEL=300 \
      -DUA_ENABLE_AMALGAMATION=TRUE \
      -DUA_ENABLE_PUBSUB=TRUE \
      -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=TRUE \
      -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=TRUE \
      -DUA_ENABLE_PUBSUB_DELTAFRAMES=TRUE \
      -DUA_ENABLE_PUBSUB_MQTT=TRUE \
      -DUA_ENABLE_ENCRYPTION=TRUE \
      -DUA_MULTITHREADING 100
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: alpine:3.11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-10-31:
GCC compiler?

je1h, 2020-10-31:
```

je1h, 2020-11-03:
That's how I do reproduce the issue:

1. Run Alpine container interactively:
```bash
docker run -it --entrypoint sh alpine:3.11
```
2. Inside container install packages required for building:
```bash
apk add --no-cache libnl libnl3-dev cmake binutils make musl-dev gcc g++ openssl-dev sudo alpine-sdk linux-headers automake autoconf libtool python3
```
3. Install Mbed TLS dependancy:
```bash
cd home/
git clone https://github.com/ARMmbed/mbedtls.git
cd mbedtls/
mkdir build
cd build/
cmake ..
make install
```
4. Clone and make open62541:
```bash
cd ../..
git clone https://github.com/open62541/open62541.git
cd open62541/
git checkout v1.1.2
mkdir build
cd build/
cmake -DUA_LOGLEVEL=300 -DUA_ENABLE_AMALGAMATION=TRUE -DCMAKE_BUILD_TYPE=Release ..
make
```

Hope that makes sense.

traversaro, 2020-11-14:
In general for builds that are not in the official CI, it make sense to disable the `-Werror` flag because it is perfectly possible that a new compile comes out with new warnings that were not present when the open62541 release was done. I had the same problem packaging open62541 in [conda-forge](https://conda-forge.org/), see https://github.com/conda-forge/staged-recipes/pull/13231 .

A simple solution to disable the `-Werror` flag in *nix builds is to add: 
~~~
-DUA_ARCH_REMOVE_FLAGS="-Werror" 
~~~

je1h, 2020-11-30:
> In general for builds that are not in the official CI, it make sense to disable the `-Werror` flag because it is perfectly possible that a new compile comes out with new warnings that were not present when the open62541 release was done. I had the same problem packaging open62541 in [conda-forge](https://conda-forge.org/), see [conda-forge/staged-recipes#13231](https://github.com/conda-forge/staged-recipes/pull/13231) .
> 
> A simple solution to disable the `-Werror` flag in *nix builds is to add the CMake option:
> 
> ```
> -DUA_ARCH_REMOVE_FLAGS="-Werror" 
> ```

Thanks for the hint!

NoelGraf, 2021-04-18:
What is the status of the issue? Can the issue be closed?

jpfr, 2021-04-20:
The problem reported here is not from the compiler, but the standard library.
Alpine uses the musl libc.

@NoelGraf please add a build in the GitHub actions that uses musl.
Importantly also the include files that come with musl, not only the shared object.
It’s available in Debian since forever...

traversaro, 2021-09-01:
Related recent PR: https://github.com/open62541/open62541/pull/4572 .

closed 2025-02-04 by andreasebner