open62541/open62541 #3988: Correct way to reduce the RAM requirements of application by reducing the size of information model?

nqduy35, 2020-11-03:
Hello everyone, 

I heard about the myth that "it is possible to configure minimal servers that require less than 100kB of RAM and ROM" published [here](https://open62541.org/doc/current/building.html) 
I did everything except to reduce the information model, and the RAM and ROM requirements are still high: running a server example requires 800KB of RAM. So I decided to reduce the information model NS0 by trying to remove unnecessary nodes. 
I want to find a "correct way" to do it. One way to do it I post in the issue [#3978](https://github.com/open62541/open62541/issues/3978), but it does not work. Please give me some advice? 

Many thanks for considering my request.  

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.2-269-gef0c6873
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-02:
Have you tried selecting the minimal ns0 in the build config?

nqduy35, 2020-12-02:
I have run through and applied every tip in the tutorial. In the tutorial, there is also these lines 

> The RAM requirements of a server are mostly due to the following settings:
>  - The size of the information model 
>  - The number of connected clients
>  - The configured maximum message size that is preallocated

Follow the above remarks, I have already:
- Reduced the message size.
- Tested with only one client and one server. 

But RAM consumption is still high.

juangburgos, 2021-04-22:
Already on this topic, maybe we should start thinking on a way of abstracting the node store (including references), in such a way that it can be moved to different *backends* (as done with the historic plugin).

The case being, in [QUaServer](https://github.com/juangburgos/QUaServer#Alarms) I have been playing with alarms. In specific the  *ExclusiveLevelAlarm* has a very heavy information model, just for one instance, because includes state machines, with lots of nodes and references. So I made tests with around 1000 alarms (which is not much for an industrial application) and the server eats up to 100 megs.

Now to be fair, every node in QUaServer is also a QObject, which adds to the problem, but with I profile the memory, actually the objects that conume more memory (around 40%) are references instantiated by open62541.

Ideally, by abstracting out the node store and references, one could move all that data to (for example) an SQLite database. True this would penalize speed of data access, but would reduce dramatically the use or RAM.

Tradeoffs...

closed 2025-06-06 by teccheck