open62541/open62541 #3993: Ua Expert gets 'BadSecurityModeInsufficent' during activatesession

RFil, 2020-11-04:
Ua Expert always gets 'BadSecurityModeInsufficent' during activatesession when I try to login providing username and password

MichaelKlin, 2020-11-08:
@RFil Do you get solution?

RFil, 2020-11-09:
> 
> 
> @RFil Do you get solution?

Not yet... any help is welcome

mlang-de, 2020-11-23:
See https://forum.unified-automation.com/viewtopic.php?f=22&t=3422
If security and encryption is none + Username/Password is used the UaExpert throws this warning intentionally.

RFil, 2020-11-23:
> 
> 
> See https://forum.unified-automation.com/viewtopic.php?f=22&t=3422
> If security and encryption is none + Username/Password is used the UaExpert throws this warning intentionally.

Thanks

mlang-de, 2020-11-24:
@RFil Please close this issue.

closed 2020-11-24 by RFil