open62541/open62541 #3998: Publishing interval below 1 ms

busboom, 2020-11-07:
## Description
Good afternoon, I am trying to run the example tutorial_pubsub_publisher at a faster publishing rate than 1 ms. I observed in Wireshark that reducing the publishing interval below 1 ms does not further increase the rate of network messages.

This appears to be related to the main loop of the server which listens to the port with a timeout of at least 1 ms in each iteration (libwebsockets).

Is there any way I can further speed up the publisher?

Used CMake options:
UA_ENABLE_PUBSUB ON

## Checklist
 - [x] open62541 Version (release number or git tag): v1.1.2
 - [x] Other OPC UA SDKs used (client or server): none, just Wireshark to observe network traffic
 - [X] Operating system: Yocto embedded Linux distribution
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

opcua-tsn-team-kalycito, 2020-11-08:
There is a custom callback option that allows the user to create their own callback mechanism. Using this, you can achieve a lower cycle time below 1ms with respect to your system capabilities.

Have a look at the pubsub_TSN_publisher.c example that uses threads to publish the packets at  250us cycle time.

jpfr, 2020-12-01:
For PubSub or client/server subscriptions?

busboom, 2020-12-02:
> For PubSub or client/server subscriptions?

I was referring to PubSub.

In fact, I would not only be interested in fast, cyclic publishing (it seems like the reply from opcua-tsn-team-kalycito addresses this), but also in acyclic publishing, i.e. pubslihed events as opposed to published data items. Is anyone working on that front?

closed 2021-08-06 by busboom