open62541/open62541 #4003: FreeRTOS/LWIP Client: Error 203 when connecting to second Server

henrik-the, 2020-11-11:
## Description
I am using the **amalgamated** version of open62541 Stack under **FreeRTOS/LWIP** architecture. I have recently updated my code to **v1.1.2-234-ge49cb693**. 
With **v1.0.1** the following Error didn't happen:

I am trying to connect to 2 different OPCUA Servers with the client.
The connection to the first server could sucessfully be established, but if i try to connect to a second server the "DNS lookup" fails with error 203.

Error 203 is a the EAI_MEMORY error of LWIP.
In this forum I could find a very simmilar behaviour:
https://community.cypress.com/thread/6640?start=0&tstart=0
They say, that this error happens, if the `struct addrinfo` isnt freed before the next `getaddrinfo()` call.
In the commit history i could find, that there were some changes with the handling of `freeaddrinfo()`, 
but i couldn't find a bug.

This is the output log while connecting:

```
TestClient: Connecting to Server 1!
[1970-01-01 00:00:12.399 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[1970-01-01 00:00:15.182 (UTC+0000)] info/channel       Connection 3 | SecureChannel 20 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised            lifetime of 600.00s
[1970-01-01 00:00:15.195 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[1970-01-01 00:00:15.212 (UTC+0000)] info/client        Selected Endpoint opc.tcp://192.168.1.126:4840 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[1970-01-01 00:00:15.224 (UTC+0000)] info/client        Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/Security           Policy#None
[1970-01-01 00:00:15.258 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[1970-01-01 00:00:15.280 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
connected sucessfully to server1!

TestClient: Connecting to Server 2!
[1970-01-01 00:00:17.290 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[1970-01-01 00:00:17.299 (UTC+0000)] warn/network       DNS lookup of 192.168.1.127 failed with error 203 -
[1970-01-01 00:00:17.308 (UTC+0000)] warn/client        Could not open a TCP connection to opc.tcp://192.168.1.127:4840
[1970-01-01 00:00:17.318 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
connect retval: 80ae0000
```

## Background Information / Reproduction Steps
here is the client connection code i have used to reproduce this Error:
```
void opcua_clientTest(void *arg){
    vTaskDelay(10000);
    printf("\nTestClient: Connecting to Server 1!\n");

    UA_Client *client1 = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client1));

    UA_StatusCode retval = UA_Client_connect(client1, "opc.tcp://192.168.1.126:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client1);
        printf("connect retval: %x\n",(int)retval);
    }
    if(retval == UA_STATUSCODE_GOOD) {
    	printf("connected sucessfully to server1!\n");
    }

    vTaskDelay(2000);
    printf("\nTestClient: Connecting to Server 2!\n");

    UA_Client *client2 = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client2));
    
    retval = UA_Client_connect(client2, "opc.tcp://192.168.1.127:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client2);
        printf("connect retval: %x\n",(int)retval);
    }
    if(retval == UA_STATUSCODE_GOOD) {
    	printf("connected sucessfully to server2!\n");
    }
}
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: FreeRTOS/LWIP
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

cochicde, 2020-11-16:
the equivalent to **Have you try turning it off and on again?** for problems in freeRTOS/LwIP is **Do you have enough memory?**

EAI_MEMORY appears when you run out of memory, so it supports the idea. the freeaddrinfo problem doesn't seem to be the cause, since in that function, the struct is allocated inside TCPClientConnection.

I would check first that is not a memory problem as mentioned in the famous #2511

NoelGraf, 2021-04-18:
What is the status of the issue? Can the issue be closed?

henrik-the, 2021-04-19:
I did find a fix, but i don't think it is a good one...
I added:
`freeaddrinfo(tcpClientConnection->server);`
in this line:
https://github.com/open62541/open62541/blob/ae74dc0071cc58cf96abeeaf7b9c6c39dd2ed286/arch/network_tcp.c#L881

But this codeline allowed me to connect to multiple servers.