open62541/open62541 #4005: browsing issue

fu6xu4fup6, 2020-11-13:
## Description

i used client browsing service like this ,  which was someone's code on the github.

and it can get all nodes on kepserver5 but not kepserver6

under the device,it's about 20000 tags.

if I use UA_BrowseNextRequest  it gets about 1998 tags,
and if I did'nt use UA_BrowseNextRequest it only gets about 998 tags.

```
void listTreeRecursive(UA_Client *client, UA_NodeId nodeId)
{
	size_t i, j;
	/* Browse some objects */
	UA_BrowseRequest bReq;
	UA_BrowseRequest_init(&bReq);
	//bReq.requestedMaxReferencesPerNode = 0;
	bReq.nodesToBrowse = UA_BrowseDescription_new();
	bReq.nodesToBrowseSize = 1;
	UA_NodeId_copy(&nodeId, &bReq.nodesToBrowse[0].nodeId);
	bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL;

	UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);
	UA_BrowseNextRequest bNextReq;
	UA_BrowseNextRequest_init(&bNextReq);

	// normally is set to 0, to get all the nodes, but we want to test browse next
	bNextReq.releaseContinuationPoints = UA_FALSE;
	if (&bResp.results[0].continuationPoint)
	{
		
	}
	bNextReq.continuationPoints = &bResp.results[0].continuationPoint;
	bNextReq.continuationPointsSize = 1;
	
	UA_BrowseNextResponse bNextResp = UA_Client_Service_browseNext(client, bNextReq);

	for (i = 0; i < bResp.resultsSize; i++)
	{
		for (j = 0; j < bResp.results[i].referencesSize; j++)
		{
			UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
			if ((ref->nodeClass == UA_NODECLASS_OBJECT || ref->nodeClass == UA_NODECLASS_VARIABLE || ref->nodeClass == UA_NODECLASS_METHOD)) {

				if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {
					printf("%-9d %-16d %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
						ref->nodeId.nodeId.identifier.numeric, (int)ref->browseName.name.length,
						ref->browseName.name.data, (int)ref->displayName.text.length,
						ref->displayName.text.data);

					listTreeRecursive(client, UA_NODEID_NUMERIC(ref->nodeId.nodeId.namespaceIndex,
						ref->nodeId.nodeId.identifier.numeric));
					

				}
				else if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING) {
					printf("%-9d %-16.*s %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
						(int)ref->nodeId.nodeId.identifier.string.length,
						ref->nodeId.nodeId.identifier.string.data,
						(int)ref->browseName.name.length, ref->browseName.name.data,
						(int)ref->displayName.text.length, ref->displayName.text.data);

					listTreeRecursive(client, UA_NODEID_STRING(ref->nodeId.nodeId.namespaceIndex,
						(char *)ref->nodeId.nodeId.identifier.string.data));
					
				}
			}
			/* TODO: distinguish further types */
		}
	}
	
	

	for (i = 0; i < bNextResp.resultsSize; i++)
	{
		for (j = 0; j < bNextResp.results[i].referencesSize; j++)
		{
			
			UA_ReferenceDescription *ref = &(bNextResp.results[i].references[j]);
			if ((ref->nodeClass == UA_NODECLASS_OBJECT || ref->nodeClass == UA_NODECLASS_VARIABLE || ref->nodeClass == UA_NODECLASS_METHOD)) {

				if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {
					printf("%-9d %-16d %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
						ref->nodeId.nodeId.identifier.numeric, (int)ref->browseName.name.length,
						ref->browseName.name.data, (int)ref->displayName.text.length,
						ref->displayName.text.data);

				listTreeRecursive(client, UA_NODEID_NUMERIC(ref->nodeId.nodeId.namespaceIndex,
						ref->nodeId.nodeId.identifier.numeric));

				}
				else if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING) {
					printf("%-9d %-16.*s %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
						(int)ref->nodeId.nodeId.identifier.string.length,
						ref->nodeId.nodeId.identifier.string.data,
						(int)ref->browseName.name.length, ref->browseName.name.data,
						(int)ref->displayName.text.length, ref->displayName.text.data);

				listTreeRecursive(client, UA_NODEID_STRING(ref->nodeId.nodeId.namespaceIndex,
						(char *)ref->nodeId.nodeId.identifier.string.data));
 	
				}
			}
			/* TODO: distinguish further types */
		}
	}
	UA_BrowseRequest_deleteMembers(&bReq);
	UA_BrowseResponse_deleteMembers(&bResp);
}

int main(int argc, char **argv)
{
	UA_Client *client = UA_Client_new();
	UA_ClientConfig_setDefault(UA_Client_getConfig(client));
	UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://192.168.1.100:49320");

	if (retval != UA_STATUSCODE_GOOD) {
		printf("Unable to connect!");
		UA_Client_delete(client);
		return (int)retval;
	}

	printf("Browsing nodes in objects folder:\n");
	printf("%-9s %-16s %-16s %-16s\n", "NAMESPACE", "NODEID", "BROWSE NAME", "DISPLAY NAME");
	listTreeRecursive(client, UA_NODEID_STRING(2, "Channel1.Device1"));

	printf("Program End\n");

	/* Clean up */
	std::cin.get();
	UA_Client_delete(client); /* Disconnects the client internally */
	return UA_STATUSCODE_GOOD;
}
```

[image]
[image]

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Demolishun, 2020-11-17:
I do all my nodes a numerics.  This is how I filter:
`
if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {
                if(!(ref->nodeClass & (UA_NODECLASS_REFERENCETYPE | UA_NODECLASS_OBJECTTYPE | UA_NODECLASS_DATATYPE | UA_NODECLASS_VARIABLETYPE))
                  && (ref->nodeId.nodeId.identifier.numeric < UA_NS0ID_MODELLINGRULE_MANDATORY
                  || ref->nodeId.nodeId.identifier.numeric > UA_NS0ID_MODELLINGRULE_EXPOSESITSARRAY)){
`
Hope this helps.  I don't know how I figured this out.

jpfr, 2020-12-01:
From what I see you only call BrowseNext once.
You need to call it more often until there are no further results.

closed 2020-12-19 by fu6xu4fup6

Jaibuti, 2023-10-30:
@fu6xu4fup6 请教下 我通过上述代码只获取到部分节点 没有全部获取，需要如何解决呢