open62541/open62541 #4006: Error in building the PubSub code with gcc in Ubuntu

EugenM-sie, 2020-11-13:
## Description
Hallo, i created new Project in Visual Sudio with main, some headers and sources. In Ubuntu i cloned your repository, done Cmake(i set Amalganation and Dua_Enable_Pubsub). Your example server works well. I import my project anf compile with gcc, but i get error in publish-function

## Background Information / Reproduction Steps

[image]

[image]

Used CMake options:

```bash
I set amalgination, PubSub, Discovery, Discovery_Multicast
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): current version
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04.
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Critical issue

EugenM-sie, 2020-11-13:
Hallo, @Pro , can you please give a reference or some help and can close issue

EugenM-sie, 2020-11-13:
in server_config.h i see that UA_ENABLE_PUBSUB is not defined

EugenM-sie, 2020-11-13:
I resolve it. Thanks you)

Waner98, 2020-11-13:
For people who will have the same problem in future, you cannot change the server_config.h manually but you have to add compile flag when you will recompile your library. ("cmake UA_ENABLE_PUBSUB=ON <others flags> ..")

EugenM-sie, 2020-11-13:
Hallo @Waner98 , i did not change the server_config.h manually. 
By first compiling, i set flags:
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_DISCOVERY_SEMAPHORE=ON -DUA_ENABLE_PUBSUB=ON -DUA_NAMESPACE_ZERO=FULL ..

But if i set -DUA_ENABLE_AMALGAMATION=OFF  and -DBUILD_SHARED_LIBS=OFF, it works

closed 2020-11-14 by Pro