open62541/open62541 #4007: PubSub --> Subscriber, real subscription or filter?

Waner98, 2020-11-13:
## Description
Hello,

I worked a lot on the open62541 stack but I felt on an issue on which I am not sure about my understanding.
As concern the last API that we can use for a subscriber in a PubSub context (e g: examples/pubsub_realtime/pubsub_TSN_publisher.c or examples/pubsub/tutorial_pubsub_subscribe.c), do we really have a publish/subscribe interaction between both systems ?

The subscriber seems to receive all the data and only filter the ones that it wants, but do not "subscribe" to the publisher. Am I right on this or did I miss something in the stack ?

Thx :) 
 
Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_DELTAFRAMES=OFF -DUA_ENABLE_PUBSUB_FILE_CONFIG=OFF -DUA_ENABLE_PUBSUB_ETH_UADP_ETF=OFF -DUA_ENABLE_PUBSUB_ETH_UADP_XDP=OFF -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag): 2.17.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Waner98, 2020-11-19:
@Pro do you have the time to put this into the light please? :) 
(sorry for the up, it will help me a lot to continue on my project)

o6-nbeier, 2026-10-20:
Thanks for the question and for taking the time to look into the stack so closely.

Your understanding is correct: UDP/Ethernet UADP PubSub is brokerless by design, so the subscriber never registers with the publisher and instead receives via its multicast/interface membership and filters locally with its DataSetReader (publisherId, writerGroupId, dataSetWriterId).