open62541/open62541 #4014: PublishingInterval, SamplingInterval and QueueSize of MonitoredItem

fijo1015, 2020-11-16:
Hello everybody,

I am new with OPC UA. I am using the open62541 SDK for my OPC UA Server including the information model. I want to subscribe some nodes (with the OPC Publisher of Microsoft) to publish them. For the subscription i chose a publishing interval of 1000ms and a sampling interval of 300ms. The issue is, that many sampled values are not published. I think it's because of the queue size of the monitored item.  Is that possible? If so, what setting do i have to change in the open62541 SDK?

Thank you for your help.
jo 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [x] Other OPC UA SDKs used (client or server): OPC Publisher by Microsoft
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-02:
Do you want to see updates when the variable has not changed?
You can select the filter criteria to include the current time stamp.

fijo1015, 2020-12-04:
No. I want to publish each sampled node value. Currently (sampling interval = 300ms, publishing interval = 1000ms) I do not get every sampled value. I only get the newest node value. The OPC UA Specification Part 4 says that:

> If the queue size is one, the queue becomes a buffer that always contains the newest Notification.
> In this case, if the sampling interval of the MonitoredItem is faster than the publishing interval of the
> Subscription, the MonitoredItem will be over sampling and the Client will always receive the most
> up-to-date value. The discard policy is ignored if the queue size is one.

For this reason I think the queue size is 1. But I want the following:

> On the other hand, the Client may want to subscribe to a continuous stream of Notifications without
> any gaps, but does not want them reported at the sampling interval. In this case, the MonitoredItem
> would be created with a queue size large enough to hold all Notifications generated between two
> consecutive publishing cycles. Then, at each publishing cycle, the Subscription would send all
> Notifications queued for the MonitoredItem to the Client.

I hope you understand my topic.

jpfr, 2021-09-23:
Just increase the queue size.
That should help.

closed 2021-09-23 by jpfr