open62541/open62541 #4018: build tutorial_server_alarms_conditions.c

viktor-morin, 2020-11-18:
## Description
Building example: [tutorial_server_alarms_conditions.c](https://github.com/open62541/open62541/blob/master/examples/tutorial_server_alarms_conditions.c)

## Background Information / Reproduction Steps
When building the example, I get several warning and 4 errors:

```shell
alarm_server.c: In function ‘addCondition_1’:
alarm_server.c:80:14: warning: implicit declaration of function ‘UA_Server_createCondition’; did you mean ‘UA_Server_removeConnection’? [-Wimplicit-function-declaration]
     retval = UA_Server_createCondition(server,
              ^~~~~~~~~~~~~~~~~~~~~~~~~
              UA_Server_removeConnection
alarm_server.c: In function ‘afterWriteCallbackVariable_1’:
alarm_server.c:181:19: warning: implicit declaration of function ‘UA_Server_setConditionVariableFieldProperty’; did you mean ‘UA_Server_readObjectProperty’? [-Wimplicit-function-declaration]
         retval |= UA_Server_setConditionVariableFieldProperty(server, conditionInstance_1,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   UA_Server_readObjectProperty
alarm_server.c:208:18: warning: implicit declaration of function ‘UA_Server_triggerConditionEvent’; did you mean ‘UA_Server_triggerEvent’? [-Wimplicit-function-declaration]
         retval = UA_Server_triggerConditionEvent(server, conditionInstance_1,
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                  UA_Server_triggerEvent
alarm_server.c: In function ‘afterWriteCallbackVariable_3’:
alarm_server.c:298:14: warning: implicit declaration of function ‘UA_Server_setConditionField’; did you mean ‘UA_Server_setNodeContext’? [-Wimplicit-function-declaration]
     retval = UA_Server_setConditionField(server, conditionInstance_1,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
              UA_Server_setNodeContext
alarm_server.c: In function ‘setUpEnvironment’:
alarm_server.c:441:5: error: unknown type name ‘UA_TwoStateVariableChangeCallback’; did you mean ‘UA_ExternalValueCallback’?
     UA_TwoStateVariableChangeCallback userSpecificCallback = enteringEnabledStateCallback;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     UA_ExternalValueCallback
alarm_server.c:441:62: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
     UA_TwoStateVariableChangeCallback userSpecificCallback = enteringEnabledStateCallback;
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
alarm_server.c:442:14: warning: implicit declaration of function ‘UA_Server_setConditionTwoStateVariableCallback’; did you mean ‘UA_Server_setVariableNode_valueCallback’? [-Wimplicit-function-declaration]
     retval = UA_Server_setConditionTwoStateVariableCallback(server, conditionInstance_1,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              UA_Server_setVariableNode_valueCallback
alarm_server.c:445:61: error: ‘UA_ENTERING_ENABLEDSTATE’ undeclared (first use in this function); did you mean ‘UA_NS0ID_SERVERSTATE’?
                                                             UA_ENTERING_ENABLEDSTATE);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~
                                                             UA_NS0ID_SERVERSTATE
alarm_server.c:445:61: note: each undeclared identifier is reported only once for each function it appears in
alarm_server.c:453:26: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
     userSpecificCallback = enteringAckedStateCallback;
                          ^
alarm_server.c:457:61: error: ‘UA_ENTERING_ACKEDSTATE’ undeclared (first use in this function); did you mean ‘UA_ENTERING_ENABLEDSTATE’?
                                                             UA_ENTERING_ACKEDSTATE);
                                                             ^~~~~~~~~~~~~~~~~~~~~~
                                                             UA_ENTERING_ENABLEDSTATE
alarm_server.c:465:26: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
     userSpecificCallback = enteringConfirmedStateCallback;
                          ^
alarm_server.c:469:61: error: ‘UA_ENTERING_CONFIRMEDSTATE’ undeclared (first use in this function); did you mean ‘UA_ENTERING_ENABLEDSTATE’?
                                                             UA_ENTERING_CONFIRMEDSTATE);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                             UA_ENTERING_ENABLEDSTATE
```

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): b34cde0 
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux (WSL)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-11-18:
https://github.com/open62541/open62541/blob/b34cde0fd531c51fbb3903ec07f74cf433af027b/CMakeLists.txt#L138

viktor-morin, 2020-11-19:
> https://github.com/open62541/open62541/blob/b34cde0fd531c51fbb3903ec07f74cf433af027b/CMakeLists.txt#L138

Thanks!

build using following arguments:

> cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS=ON ..

closed 2020-11-19 by viktor-morin

oriolorra, 2022-07-11:
Hi,

Is there anywhere the 'tutorial_client_alarms_conditions.c'? I mean a client to use, as example, with 'tutorial_server_alarms_conditions.c'. If you have it, could you share it?

Thanks,

viktor-morin, 2022-07-24:
> Hi,
> 
> Is there anywhere the 'tutorial_client_alarms_conditions.c'? I mean a client to use, as example, with 'tutorial_server_alarms_conditions.c'. If you have it, could you share it?
> 
> Thanks,

I haven't work with OPCUA the during the last years and have any code available, sorry.