open62541/open62541 #4019: missing <open62541/config.h>

jackybek, 2020-11-18:
## Description : Compiling error: fatal error : open62541/config.h: No such file of directory : #include <open62541/config.h>

## Background Information / Reproduction Steps
i just downloaded v1.1.3 and tried to compile my code with a statement : #include <open62541/types.h>
it complain : No such file or directory open62541/config.h

Used CMake options:

pi@OPCUA1:~/open62541 $ cmake -LACMake Warning:  No source or binary directory provided.  Both will be assumed to be the  same as the current working directory, but note that this warning will  become a fatal error in future CMake releases.-- open62541 Version: 1.1.3-- The selected architecture is: posix-- Test CC flag -std=c99-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -pipe-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wall-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wextra-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wpedantic-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Werror-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wno-static-in-inline-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wno-overlength-strings-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wno-unused-parameter-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wc++-compat-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wformat-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wformat-security-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wformat-nonliteral-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wmissing-prototypes-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wstrict-prototypes-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wredundant-decls-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wuninitialized-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Winit-self-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wcast-qual-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wstrict-overflow-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wnested-externs-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wmultichar-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -Wundef-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -fno-strict-aliasing-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -fexceptions-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -fstack-protector-strong-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -fstack-clash-protection-- Performing Test flag_supported-- Performing Test flag_supported - Success-- Test CC flag -mcet-- Performing Test flag_supported-- Performing Test flag_supported - Failed-- Test CC flag -fcf-protection-- Performing Test flag_supported-- Performing Test flag_supported - Failed-- Could NOT find LATEX (missing: LATEX_COMPILER)-- Configuring done-- Generating done-- Build files have been written to: /home/pi/open62541-- Cache valuesBIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUNDBIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUNDBUILD_SHARED_LIBS:BOOL=OFFCLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUNDCLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUNDCLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUNDCMAKE_AR:FILEPATH=/usr/bin/arCMAKE_BUILD_TYPE:STRING=DebugCMAKE_COLOR_MAKEFILE:BOOL=ONCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8CMAKE_CXX_FLAGS:STRING=CMAKE_CXX_FLAGS_DEBUG:STRING=-gCMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUGCMAKE_CXX_FLAGS_RELEASE:STRING=-O3 -DNDEBUGCMAKE_CXX_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUGCMAKE_C_COMPILER:FILEPATH=/usr/bin/ccCMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8CMAKE_C_FLAGS:STRING=CMAKE_C_FLAGS_DEBUG:STRING=-gCMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUGCMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUGCMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUGCMAKE_EXE_LINKER_FLAGS:STRING=CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFFCMAKE_INSTALL_BINDIR:PATH=binCMAKE_INSTALL_DATADIR:PATH=CMAKE_INSTALL_DATAROOTDIR:PATH=shareCMAKE_INSTALL_DOCDIR:PATH=CMAKE_INSTALL_INCLUDEDIR:PATH=includeCMAKE_INSTALL_INFODIR:PATH=CMAKE_INSTALL_LIBDIR:PATH=libCMAKE_INSTALL_LIBEXECDIR:PATH=libexecCMAKE_INSTALL_LOCALEDIR:PATH=CMAKE_INSTALL_LOCALSTATEDIR:PATH=varCMAKE_INSTALL_MANDIR:PATH=CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/includeCMAKE_INSTALL_PREFIX:PATH=/usr/localCMAKE_INSTALL_RUNSTATEDIR:PATH=CMAKE_INSTALL_SBINDIR:PATH=sbinCMAKE_INSTALL_SHAREDSTATEDIR:PATH=comCMAKE_INSTALL_SYSCONFDIR:PATH=etcCMAKE_LINKER:FILEPATH=/usr/bin/ldCMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/makeCMAKE_MODULE_LINKER_FLAGS:STRING=CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=CMAKE_NM:FILEPATH=/usr/bin/nmCMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopyCMAKE_OBJDUMP:FILEPATH=/usr/bin/objdumpCMAKE_RANLIB:FILEPATH=/usr/bin/ranlibCMAKE_SHARED_LINKER_FLAGS:STRING=CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=CMAKE_SKIP_INSTALL_RPATH:BOOL=NOCMAKE_SKIP_RPATH:BOOL=NOCMAKE_STATIC_LINKER_FLAGS:STRING=CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=CMAKE_STRIP:FILEPATH=/usr/bin/stripCMAKE_VERBOSE_MAKEFILE:BOOL=FALSECPACK_SOURCE_RPM:BOOL=OFFCPACK_SOURCE_TBZ2:BOOL=ONCPACK_SOURCE_TGZ:BOOL=ONCPACK_SOURCE_TXZ:BOOL=ONCPACK_SOURCE_TZ:BOOL=ONCPACK_SOURCE_ZIP:BOOL=OFFDVIPDF_CONVERTER:FILEPATH=DVIPDF_CONVERTER-NOTFOUNDDVIPS_CONVERTER:FILEPATH=DVIPS_CONVERTER-NOTFOUNDGIT_EXECUTABLE:FILEPATH=/usr/bin/gitHTLATEX_COMPILER:FILEPATH=HTLATEX_COMPILER-NOTFOUNDLATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUNDLATEX_COMPILER:FILEPATH=LATEX_COMPILER-NOTFOUNDLUALATEX_COMPILER:FILEPATH=LUALATEX_COMPILER-NOTFOUNDMAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUNDPDFLATEX_COMPILER:FILEPATH=PDFLATEX_COMPILER-NOTFOUNDPDFTOPS_CONVERTER:FILEPATH=PDFTOPS_CONVERTER-NOTFOUNDPS2PDF_CONVERTER:FILEPATH=PS2PDF_CONVERTER-NOTFOUNDPYTHON_EXECUTABLE:FILEPATH=/usr/bin/pythonSPHINX_EXECUTABLE:FILEPATH=/usr/bin/sphinx-buildUA_AMALGAMATION_ARCHITECTURES:STRING=UA_ARCHITECTURE:STRING=posixUA_ARCH_ADD_FLAGS:STRING=UA_ARCH_EXTRA_INCLUDES:STRING=UA_ARCH_LINKER_FLAGS:STRING=UA_ARCH_REMOVE_FLAGS:STRING=UA_BUILD_EXAMPLES:BOOL=OFFUA_BUILD_FUZZING:BOOL=OFFUA_BUILD_FUZZING_CORPUS:BOOL=OFFUA_BUILD_OSS_FUZZ:BOOL=OFFUA_BUILD_TOOLS:BOOL=OFFUA_BUILD_UNIT_TESTS:BOOL=OFFUA_DEBUG:BOOL=OFFUA_DEBUG_DUMP_PKGS:BOOL=OFFUA_ENABLE_AMALGAMATION:BOOL=OFFUA_ENABLE_COVERAGE:BOOL=OFFUA_ENABLE_DA:BOOL=ONUA_ENABLE_DETERMINISTIC_RNG:BOOL=OFFUA_ENABLE_DISCOVERY:BOOL=OFFUA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFFUA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ONUA_ENABLE_ENCRYPTION:BOOL=OFFUA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFFUA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFFUA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFFUA_ENABLE_HARDENING:BOOL=ONUA_ENABLE_HISTORIZING:BOOL=OFFUA_ENABLE_IMMUTABLE_NODES:BOOL=OFFUA_ENABLE_JSON_ENCODING:BOOL=OFFUA_ENABLE_MALLOC_SINGLETON:BOOL=OFFUA_ENABLE_METHODCALLS:BOOL=ONUA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFFUA_ENABLE_NODEMANAGEMENT:BOOL=ONUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS: [... cut ...] d |   |  
-- | -- | -- | -- | --
  | ReplyForward

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.1.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-11-19:
The file open62541/config.h is created in your build folder.
That folder should be added automatically to the search path.

Something like
<build>/src_generated/include/open6251/config.h

jackybek, 2020-11-19:
thanks Julius.
I found the file.

Now when i build using the following:
gcc -std=c99 -v -I/usr/include/ -I/usr/include/libxml2/ -I/home/pi/open62541/ -I/home/pi/open62541/plugins/ -lpthread -lopen62541 -lxml2 myNewMethod.c myNewServer.c -o myNewServer

during the link process, it complain:( i already check the symbolic links are present in /usr/lib/ : libopen62541.so -> libopen62541.so.1->libopen62541.so.1.1.3)  What other libraries am i missing?

/usr/bin/ld: /tmp/ccNXsMl4.o: in function `StartOPCUAServer':
myNewServer.c:(.text+0x5b8): **undefined reference** to `UA_ServerConfig_setMinimal'
/usr/bin/ld: myNewServer.c:(.text+0x5d8): **undefined reference** to `UA_ServerConfig_setDefault'
/usr/bin/ld: /tmp/ccNXsMl4.o: in function `run_UA_Client':
myNewServer.c:(.text+0x170c): undefined reference to `UA_Client_writeValueAttribute'
/usr/bin/ld: myNewServer.c:(.text+0x17b0): undefined reference to `UA_Client_writeValueAttribute'
/usr/bin/ld: myNewServer.c:(.text+0x1850): undefined reference to `UA_Client_writeValueAttribute'
/usr/bin/ld: myNewServer.c:(.text+0x18f4): undefined reference to `UA_Client_writeValueAttribute'
/usr/bin/ld: myNewServer.c:(.text+0x1994): undefined reference to `UA_Client_writeValueAttribute'
/usr/bin/ld: /tmp/ccNXsMl4.o:myNewServer.c:(.text+0x1a38): more undefined references to `UA_Client_writeValueAttribute' follow
/usr/bin/ld: /tmp/ccNXsMl4.o: in function `run_UA_Client':
myNewServer.c:(.text+0x2a7c): undefined reference to `UA_Client_readValueAttribute'
/usr/bin/ld: myNewServer.c:(.text+0x3338): **undefined reference** to `UA_Client_writeValueAttribute'
/usr/bin/ld: myNewServer.c:(.text+0x3398): undefined reference to `UA_Client_writeValueAttribute'
/usr/bin/ld: /tmp/ccNXsMl4.o: in function `main':
myNewServer.c:(.text+0xa770): undefined reference to `UA_ServerConfig_setDefault'

prater27, 2021-01-14:
I would recommend in general building with CMake files, otherwise it can be hard. I leave here a "template" I use in case someone in the future has a similar problem, I think someone could find it useful. This template is also "prepared" for compiling custom information models in the server given a custom xml nodeset file (for example generated from a xml model design file).

```CMake
cmake_minimum_required(VERSION 3.0...3.12)
if(${CMAKE_VERSION} VERSION_LESS 3.12)
    cmake_policy(VERSION ${CMAKE_MAJOR_VERSION}.${CMAKE_MINOR_VERSION})
endif()

#set(CMAKE_CXX_STANDARD 17)
#set(CMAKE_CXX_STANDARD_REQUIRED ON)
#set(CMAKE_CXX_EXTENSIONS OFF)

project(customPublisher_unionsTest)
# open62541 v1.2 must be installed!
# If in custom path, then use -DCMAKE_PREFIX_PATH=/home/user/installcmake_minimum_required(VERSION 3.2)
find_package(open62541 1.2 REQUIRED COMPONENTS FullNamespace)

#set(CMAKE_C_COMPILER "g++")
#set(CMAKE_CXX_COMPILER "g++")

SET_SOURCE_FILES_PROPERTIES( main_SubscribeUADP.c PROPERTIES LANGUAGE CXX )

set(GENERATE_OUTPUT_DIR "${CMAKE_BINARY_DIR}/src_generated/")
include_directories("${GENERATE_OUTPUT_DIR}")
file(MAKE_DIRECTORY "${GENERATE_OUTPUT_DIR}")

include_directories("${PROJECT_SOURCE_DIR}")
include_directories("${PROJECT_SOURCE_DIR}/OPCUA_ABCD/pubsub")
include_directories("${PROJECT_SOURCE_DIR}/OPCUA_ABCD/deps")
include_directories("${PROJECT_SOURCE_DIR}/OPCUA_ABCD/include")
include_directories("${PROJECT_SOURCE_DIR}/OPCUA_ABCD/include/open62541/plugin")
include_directories("${PROJECT_SOURCE_DIR}/OPCUA_ABCD/plugins/")

# The generated files are: "namespace_${NAMESPACE_NAME}_... .c/.h"
set(NAMESPACE_NAME "abcd_opcua")
# The compiled namespace is: "${NAMESPACE_PREFIX}-ns-${NAMESPACE_NAME}"
set(NAMESPACE_PREFIX "pref")
#add_subdirectory("${PROJECT_SOURCE_DIR}/OPCUA_ABCD")

ua_generate_nodeset_and_datatypes(
        INTERNAL
        NAME "${NAMESPACE_NAME}"
        FILE_NS "${CMAKE_CURRENT_LIST_DIR}/OPCUA_ABCD/model/Published/ABCD.NodeSet2.xml"
        TARGET_PREFIX "${NAMESPACE_PREFIX}"
        FILE_CSV "${CMAKE_CURRENT_LIST_DIR}/OPCUA_ABCD/model/Published/ABCD_OPCUA.csv"
        FILE_BSD "${CMAKE_CURRENT_LIST_DIR}/OPCUA_ABCD/model/Published/ABCD.Types.bsd"
        OUTPUT_DIR "${GENERATE_OUTPUT_DIR}"
    # This namespace index must match the order in which you are adding the nodeset in the source code
        NAMESPACE_MAP 1:https://abcd.org
)

add_executable(${PROJECT_NAME}
               ${UA_NODESET_ABCD_OPCUA_SOURCES}
               ${UA_TYPES_ABCD_OPCUA_SOURCES}
               "customPublish.c")

# Make sure the nodeset compiler is execute before compiling the main file
add_dependencies(${PROJECT_NAME}
                 ${NAMESPACE_PREFIX}-ns-${NAMESPACE_NAME}
                 )

target_link_libraries(${PROJECT_NAME} open62541::open62541)
```

NoelGraf, 2021-04-18:
What is the status of the issue? Can the issue be closed?

NoelGraf, 2021-04-29:
I would close the issue for now.

closed 2021-04-29 by NoelGraf