open62541/open62541 #4021: UAExpert error while Connecting to simple Server

rass92, 2020-11-19:
## Description

Hello Community,

I am making a simple OPCUA Server and I am having problems to connect with the UAExpert Client. I am facing with all the examples from the open6241 examples, but I am trying the simple server and still happens

When I try to connect this is the error that I am facing "... failed (BadServiceUnsupported)..."
[image]

Here you can see the wireshark captures
[image]

This is the code of the simple Server.

#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

This is the terminal messages

[image]

## Background Information / Reproduction Steps

examples that I have tried

Simple server from open62541
tutorial_pubsub_publish
tutorial_server_firststeps
 

I am using Linux Mint

.$ cat /etc/os-release 
NAME="Linux Mint"
VERSION="19.3 (Tricia)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.3"
VERSION_ID="19.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tricia
UBUNTU_CODENAME=bionic

Used CMake options:

```bash
I have try with many examples compile with the following cmake flags
- UA_BUILD_EXAMPLES
- UA_ENABLE_PUBSUB
- UA_ENABLE_PUBSUB_ETH_UADP
- UA_ENABLE_PUBSUB_INFORMATIONMODEL
- UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS
- UA_NAMESPACE_ZERO 		FULL
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

prater27, 2020-11-23:
I think your problem might have to do with the way you are using UA Expert. 
Try to directly specify the endpoint of the server: click in the + button for adding a server, give it a name, go to the "advanced" tab and directly write there the Endpoint of the server (opc.tcp://xxx.yyy.z.k:port, for example opc.tcp://localhost:4840 )

rass92, 2020-11-24:
dear @prater27. Thank you for your answer.

Yes I am able to connect to the server with the address that appears when you run the server

[2020-11-24 11:26:58.975 (UTC+0100)] info/network TCP network layer listening on opc.tcp://localhost:4840

Thank you for your help

closed 2020-11-24 by rass92