open62541/open62541 #4022: Using OPC UA server on TI RM48

Hben580120, 2020-11-20:
i just start to learn about OPC UA and want to porting server on my RM48 board which use keil C compiler and run lwip without OS. 
My problem is that i don't know how to start and where should i start.
there is a lot of examples here but i couldn't find something for the embedded systems which use lwip without OS. i just want to read and write some variables using the server.
so can any one help me and let me know where i can start here?

johndzilva, 2020-11-23:
open62541 has a support in a freeRTOS+lwip environment. 
https://github.com/open62541/open62541/tree/master/arch/freertosLWIP

If you don't want to use OS you may need do some changes in the mentioned files above. 
Start with these example 
Server: https://github.com/open62541/open62541/blob/master/examples/tutorial_server_firststeps.c
Client: https://github.com/open62541/open62541/blob/master/examples/tutorial_client_firststeps.c

Hben580120, 2020-11-23:
I tried to download from github and found open62541.c/.h according to the method provided on the website
But it cannot be compiled successfully in the environment of Keil C
Because the cmake method provided by the website has OS
Is there any cmake without OS that can make open62541.c/.h compile successfully under Keil C?

> open62541 has a support in a freeRTOS+lwip environment.
> https://github.com/open62541/open62541/tree/master/arch/freertosLWIP
>
> If you don't want to use OS you may need do some changes in the [...]

andreasebner, 2025-07-30:
Since the new event loop landed, the arch plugins required a lot of rework, so there's a new plugin for lwip now

closed 2025-07-30 by andreasebner