open62541/open62541 #4023: Can not Access to  Objects with Unified Automation Uaexpert(compiled for ARM-Arch)

EugenM-sie, 2020-11-20:
## Description
I add some objets and variables in Server and I can not  see Objects with Unified Automation Uaexpert. I compiled server for ARM-Architecture and execute it on Linux, The  UA client i call on Windows. All Other Configuratiion of Server I can see
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): least release
 - [x] Other OPC UA SDKs used (client or server): UA gui Client
 - [x] Operating system: Windows, Debian 9
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

EugenM-sie, 2020-11-20:
Hallo @pro Do you have the same Issue. Or can can you maybe reproduce error?

closed 2020-11-23 by EugenM-sie

viktor-morin, 2020-11-24:
@EugenM-sie 
I downloaded the latest package, get same error.
How did you solve it?