open62541/open62541 #4029: Change variable on Server from client, but no setVariableNode_valueCallback Funktion

EugenM-sie, 2020-11-25:
## Description
Hallo, I want to change variable on Server from client, but there is no setVariableNode_valueCallback Funktion like on the server side. I want to change node continuously 
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

EugenM-sie, 2020-11-25:
Hallo, @jpfr, sorry for my question. I want to change node continuously on my Server, but from the Client. There are some Funktions or Threads

o6-nbeier, 2026-10-20:
Thanks for opening this issue.

onRead/onWrite are server-side value hooks with no client counterpart, but you can write to the node continuously from the client by registering UA_Client_addRepeatedCallback and calling UA_Client_writeValueAttribute (or the _async variant) from it while driving UA_Client_run_iterate.