open62541/open62541 #4037: Server hangs on securely connecting after system boot

JohannesKauffmann, 2020-11-30:
## Description
Hi,

We are facing an issue where the server hangs and stops sending packets if an attempt to connect was made within two minutes of the system booting. This only happens when the server is configured with a self-signed certificate and the #Basic256Sha256 security policy. When waiting for several minutes (more than 5), a connection can be established but only if no previous attempt to connect was made since system bootup. Connecting right away with the #None security policy is succesfull, but successive attempts to connect securely fail.

## Background Information / Reproduction Steps
A minimal repro is attached below and can be compiled like any of the other examples. The used certificate and key are generated with the default options from the included python script. The attached log is also taken from the minimal repro.

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release -DUA_NAMESPACE_ZERO=REDUCED -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ARCH_REMOVE_FLAGS="-Werror" -DUA_LOGLEVEL=100 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): tag v1.1.2
 - [x] Other OPC UA SDKs used (client or server): UAExpert client 1.5.1 331
 - [x] Operating system: Poky Linux 2.6.1 for server, Windows 10 2004 for UAExpert
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: 
[loglevel100.txt](https://github.com/open62541/open62541/files/5615033/loglevel100.txt)

 - [x] Wireshark network dump attached: 
[server_hang.zip](https://github.com/open62541/open62541/files/5614533/server_hang.zip)

 - [x] Self-contained code example attached: 
[hang-repro.zip](https://github.com/open62541/open62541/files/5615006/hang-repro.zip)

 - [x] Critical issue

jpfr, 2020-12-01:
For our convenience, can you run up to the hang in a debugger and post a stacktrace?

Also, do you have multithreading enabled?

JohannesKauffmann, 2020-12-01:
Thank you for your repsonse.

How would suggest I run the program in a debugger, keeping in mind it is automatically started at boot time by an init.d script? I have no clue as to how one would accomplish that, but I'd be interested to hear your thoughts on such techniques.

The library was built with the options specified above and the multithreading option was left untouched.

jpfr, 2020-12-02:
.you can attach debuggers to a running process.
On Linux, just find the process id (pid).

JohannesKauffmann, 2020-12-03:
Ah, right. Attaching to the program after the fact. I have attached the stack trace below.

[stacktrace.txt](https://github.com/open62541/open62541/files/5634557/stacktrace.txt)

schroeder-, 2020-12-03:
Maybe you are running out of entropy from /dev/random and the system halt your programm until it has enough entropy. I remember some issues with booting for other linux applications linked to entropy sometime ago.

JohannesKauffmann, 2020-12-04:
Well, looking at the stack trace, one of the last calls is to `getentropy()`, which, according to the man page, indeed blocks. The system in question is a small, embedded device with not much activity going on to generate randomness. Do you reckon there is anything we could do about that?

JohannesKauffmann, 2020-12-08:
After further testing with a physical keyboard and mouse, we confirmed the issue is indeed entropy starvation. Running the [haveged](issihosts.com/haveged/) daemon on startup mitigates this problem.

closed 2020-12-08 by JohannesKauffmann