open62541/open62541 #4042: Encoding of generated Unions and padding

basyskom-jvoe, 2020-12-02:
I have modelled the following type:
```
    <opc:StructuredType BaseType="ua:Union" Name="MyUnionType">
        <opc:Field TypeName="opc:UInt32" Name="SwitchField"/>
        <opc:Field SwitchField="SwitchField" TypeName="opc:Float" SwitchValue="1" Name="FloatValue"/>
        <opc:Field SwitchField="SwitchField" TypeName="opc:CharArray" SwitchValue="2" Name="StringValue"/>
    </opc:StructuredType>
```
The nodeset compiler generates an enumeration, and a struct containing the switch field an a union of float and string.
The binary encoder fails to encode the float member and always inserts 0 while the string member is correctly encoded.

A look at the generated type description seems to explain the behavior:
```
static UA_DataTypeMember MyUnionType_members[2] = {
{
    UA_TYPENAME("FloatValue") /* .memberName */
    UA_TYPES_FLOAT, /* .memberTypeIndex */
    0, /* .padding */
    true, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},
{
    UA_TYPENAME("StringValue") /* .memberName */
    UA_TYPES_STRING, /* .memberTypeIndex */
    sizeof(UA_UInt32), /* .padding */
    true, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},};
```
According to my debugger, there are 4 bytes of padding between the switch field and the union. The string member is defined as having a padding of 32 bytes which compensates the padding inside the struct while the access to the float member always reads the four padding bytes.
```
static status
encodeBinaryUnion(const void *src, const UA_DataType *type, Ctx *ctx) {
...
    uintptr_t ptr = ((uintptr_t)src) + UA_TYPES[UA_TYPES_UINT32].memSize + m->padding;
...
```
Please provide the following information:

 - [X ] open62541 Version (release number or git tag): 1.1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: Ubuntu 16.04, GCC 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-02:
@andreasebner can you have a lok.?
Unions...

prater27, 2020-12-15:
I think I found an error maybe connected to this one in the generation of code for the union. 
In the DataType added to the DataTypes array, the /* .membersSize */ I think is one more than it should (compare to the union code example given in the custom datatype example). 
So for example, for the particular case reported in this issue:
```C++
/* UnionTestDataType */
{
    {2, UA_NODEIDTYPE_NUMERIC, {15152LU}}, /* .typeId */
    {2, UA_NODEIDTYPE_NUMERIC, {15239LU}}, /* .binaryEncodingId */
    sizeof(UA_UnionTestDataType), /* .memSize */
    UA_TYPES_CRCL_OPCUA_UNIONTESTDATATYPE, /* .typeIndex */
    UA_DATATYPEKIND_UNION, /* .typeKind */
    false, /* .pointerFree */
    false, /* .overlayable */
    3, /* .membersSize */    <----------------------------------------------It should be 2 instead of 3
    UnionTestDataType_members  /* .members */
    UA_TYPENAME("UnionTestDataType") /* .typeName */
},
```

prater27, 2020-12-16:
@jpfr  Doing the following modifications to the file "backend_open62541_typedefinitions.py" partially solves this issue. I say partially because the generated code seems right, but nevertheless I am not capable of making the unions generated work correctly.
In def print_datatype, the printed .membersSize is different depending on whether is union or not. Using an auxiliar variable for storing the .membersSize is better, but just as quick fix I used:
```python
    def print_datatype(self, datatype, namespaceMap):
        typeid = "{%s, %s}" % (namespaceMap[datatype.namespaceUri], getNodeidTypeAndId(datatype.nodeId))
        binaryEncodingId = "{%s, %s}" % (namespaceMap[datatype.namespaceUri],
                                         getNodeidTypeAndId(datatype.binaryEncodingId))
        idName = makeCIdentifier(datatype.name)
        pointerfree = "true" if datatype.pointerfree else "false"
        isUnion = isinstance(datatype, StructType) and datatype.is_union   # -------NEW BEGIN--------
        if isUnion:
            return "{\n" + \
                  "    " + typeid + ", /* .typeId */\n" + \
                  "    " + binaryEncodingId + ", /* .binaryEncodingId */\n" + \
                  "    sizeof(UA_" + idName + "), /* .memSize */\n" + \
                  "    " + self.get_type_index(datatype) + ", /* .typeIndex */\n" + \
                  "    " + self.get_type_kind(datatype) + ", /* .typeKind */\n" + \
                  "    " + pointerfree + ", /* .pointerFree */\n" + \
                  "    " + self.get_type_overlayable(datatype) + ", /* .overlayable */\n" + \
                  "    " + str(len(datatype.members)-1) + ", /* .membersSize */\n" + \  
                  "    %s_members" % idName + "  /* .members */\n" + \
                  "    UA_TYPENAME(\"%s\") /* .typeName */\n" % idName + \
                  "}"
        else:                                                 #       -------NEW END--------
            return "{\n" + \
                  "    " + typeid + ", /* .typeId */\n" + \
                  "    " + binaryEncodingId + ", /* .binaryEncodingId */\n" + \
                  "    sizeof(UA_" + idName + "), /* .memSize */\n" + \
                  "    " + self.get_type_index(datatype) + ", /* .typeIndex */\n" + \
                  "    " + self.get_type_kind(datatype) + ", /* .typeKind */\n" + \
                  "    " + pointerfree + ", /* .pointerFree */\n" + \
                  "    " + self.get_type_overlayable(datatype) + ", /* .overlayable */\n" + \
                  "    " + str(len(datatype.members)) + ", /* .membersSize */\n" + \
                  "    %s_members" % idName + "  /* .members */\n" + \
                  "    UA_TYPENAME(\"%s\") /* .typeName */\n" % idName + \
                  "}"

```
and in def print_members one additional condition must be added:
```python
...
            if len(member_name) > 0:
                member_name_capital = member_name[0].upper() + member_name[1:]
            m = "\n{\n    UA_%s_%s, /* .memberTypeIndex */\n" % (
                member.member_type.outname.upper(), makeCIdentifier(member.member_type.name.upper()))
            m += "    "
            if not before and not isUnion:      # MODIFICATION (original is "if not before")
                m += "0,"
            elif isUnion:
                m += "sizeof(UA_UInt32),"
            else:
                if member.is_array:
                    m += "offsetof(UA_%s, %sSize)" % (idName, member_name)
                else:
...
```
With the previous modifications, from the DesignFile:
```xml
    <!-- ### UnionTestDataType ###-->

    <DataType SymbolicName="CRCL:UnionTestDataType" BaseType="ua:Union" IsAbstract="false" SupportsEvents="true" ValueRank="Scalar">
        <Description>UnionTestDataType</Description>
        <Fields>
            <Field Name="switchField" DataType="ua:UInt32" ModellingRule="Mandatory" ValueRank="Scalar">
                <Description>switchField</Description>
            </Field>
            <Field Name="stringValue" DataType="ua:String" ModellingRule="Mandatory" ValueRank="Scalar">
                <Description>String</Description>
            </Field>
            <Field Name="booleanValue" DataType="ua:Boolean" ModellingRule="Mandatory" ValueRank="Scalar">
[... 41 lines cut ...]
DataTypes:
```C
...,
/* UnionTestDataType */
{
    {2, UA_NODEIDTYPE_NUMERIC, {15031LU}}, /* .typeId */
    {2, UA_NODEIDTYPE_NUMERIC, {15032LU}}, /* .binaryEncodingId */
    sizeof(UA_UnionTestDataType), /* .memSize */
    UA_TYPES_CRCL_OPCUA_UNIONTESTDATATYPE, /* .typeIndex */
    UA_DATATYPEKIND_UNION, /* .typeKind */
    false, /* .pointerFree */
    false, /* .overlayable */
    3, /* .membersSize */
    UnionTestDataType_members  /* .members */
    UA_TYPENAME("UnionTestDataType") /* .typeName */
},
```

Nevertheless, when I inspect them with UA_Expert, apparently the DataType seems right:
[image]

But when I instantiate it, all the fields are shown and it does not behave as it should (for example as the custom data type union does in the provided examples):
[image]

Any idea where else could be the error? I could try to fix it, but I would need some guidance about where to look for it...

jpfr, 2021-09-23:
Fixed via #3979.

closed 2021-09-23 by jpfr