open62541/open62541 #4043: Monitoring for large values

basyskom-jvoe, 2020-12-02:
I have an array of extension objects. Trying to create a monitored item for it ends in an assertion:
```
ua_types_encoding_binary.c:862: ExtensionObject_encodeBinary: Assertion `ret != 0x80080000' failed.
```

The reason is that detectValueChangeWithFilter() attempts encoding on the stack first and falls back to allocating a buffer on the heap if this doesn't work.
```
    /* Stack-allocate some memory for the value encoding. We might heap-allocate
     * more memory if needed. This is just enough for scalars and small
     * structures. */
    UA_STACKARRAY(UA_Byte, stackValueEncoding, UA_VALUENCODING_MAXSTACK);
    UA_ByteString valueEncoding;
    valueEncoding.data = stackValueEncoding;
    valueEncoding.length = UA_VALUENCODING_MAXSTACK;

    /* Encode the value */
    UA_Byte *bufPos = valueEncoding.data;
    const UA_Byte *bufEnd = &valueEncoding.data[valueEncoding.length];
    UA_StatusCode retval = UA_encodeBinary(value, &UA_TYPES[UA_TYPES_DATAVALUE],
                                           &bufPos, &bufEnd, NULL, NULL);
    if(retval == UA_STATUSCODE_BADENCODINGERROR) {
        size_t binsize = UA_calcSizeBinary(value, &UA_TYPES[UA_TYPES_DATAVALUE]);
        if(binsize == 0)
            return UA_STATUSCODE_BADENCODINGERROR;

        if(binsize > UA_VALUENCODING_MAXSTACK) {
            retval = UA_ByteString_allocBuffer(&valueEncoding, binsize);
            if(retval == UA_STATUSCODE_GOOD) {
                bufPos = valueEncoding.data;
                bufEnd = &valueEncoding.data[valueEncoding.length];
                retval = UA_encodeBinary(value, &UA_TYPES[UA_TYPES_DATAVALUE],
                                         &bufPos, &bufEnd, NULL, NULL);
            }
        }
    }
```
Unfortunately, the assertions that have been added in ua_types_encoding_binary.c break this mechanism.

## Checklist

Please provide the follow

 [X] open62541 Version (release number or git tag): 1.1
 [X ] Operating system:

jpfr, 2020-12-05:
Added a unit test for 1.2: #4051

This seems to have fixed since 1.1 and could not be reproduced.
The solution should happen in the binary encoding layer, not in the subscription layer.

It is important to try encoding the monitored value in the static array first.
This gives huge speed improvements for the majority of cases where the value is small and no change is detected.

The problem here is that we have to limit the "restore points" during encoding.
When we run into the end of the buffer, we send out everything until the "restore point", exchange the buffer and continue from the last "restore point".

The issue was probably fixed when we allowed any NodeIds for the data types (not only integer).

basyskom-jvoe, 2020-12-08:
I have ported my project to open62541 v1.2 and still have the same issue
```ua_types_encoding_binary.c:856: ExtensionObject_encodeBinary: Assertion `ret != 0x80080000' failed.```

jpfr, 2020-12-15:
Fixed via #4059 on the 1.2 branch.

jpfr, 2020-12-15:
Ah, I see the problem also exists for 1.1.
Reopening.

basyskom-jvoe, 2020-12-16:
@jpfr Many thanks for the fix for 1.2, I can no longer reproduce the issue.

closed 2021-09-23 by jpfr