open62541/open62541 #4048: How to secure UDP multicast PubSub ?

Waner98, 2020-12-04:
## Description
Hi,
I saw examples and documentation concerning encryption in Client/Server case. But for PubSub with UDP, I did not see anything helping me to implement a secure communication to protect my system. Do you know where could I find these information ? 

Thanks! 

Used CMake options:

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1.2-290-g509ae5e0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Waner98, 2020-12-04:
I saw in the source code that WriterGroupConfig struct has some elements concerning securityPolicy or encoding but I do not find a concrete example of how do we do (As concern the publisher, I did not take a look at subscriber for instance)

o6-nbeier, 2026-10-20:
Thank you for opening this issue. Encrypted and signed PubSub, including a subscriber example, was added in #4418, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier