open62541/open62541 #4055: Error in linking while compiling with

moradbiagooi, 2020-12-08:
## Description

I'm trying to compile open62541 library on Armbian-20.11 Buster installed on NanoPi Neo2 H5.

I'll get the compiler error while linking. I have tried it with different methods and the result was the same.

## Background Information / Reproduction Steps

The problem arises when I set '-DUA_NAMESPACE_ZERO=FULL'.
Without this flag, it compiles fast. But this flag makes the compilation very long, and it crashes.

I have set 1 G Swap and this is the error:

cc: fatal error: Killed signal terminated program lto1
compilation terminated.
lto-wrapper: fatal error: /usr/bin/cc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/open62541.dir/build.make:203: bin/libopen62541.so.1.1.2] Error 1
make[1]: *** [CMakeFiles/Makefile2:345: CMakeFiles/open62541.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

With 4 G Swap, I found out that the error was related to namespace0.

Used CMake options:

cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL  -DUA_ENABLE_ENCRYPTION=TRUE -DUA_ENABLE_ENCRYPTION_OPENSSL=TRUE  -DUA_ENABLE_ENCRYPTION_MBEDTLS=FALSE -DEncryption=True ..

## Checklist

 - [ ] open62541 Version (release number or git tag): release 1.1.3
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached

traversaro, 2020-12-08:
Are you sure that this is not simply an error due to the fact that gcc is running out of memory? You can probably verify that by monitoring the use of memory before the crash.

moradbiagooi, 2020-12-08:
I'm not sure. I tried to extend the memory by making a 4GB swap file. The error changed but it still couldn't finish the linking process. unfortunately I forgot to copy the error. I'll try to reproduce that tomorrow. Is 4GB not enough? The memory of my device is limited.