open62541/open62541 #4057: nodeset compiler is failing to compile the example.

alexlev2004, 2020-12-09:
## Description
Trying to follow the manual in order to compile an example nodeset from here:
https://open62541.org/doc/current/nodeset_compiler.html

I am failing to build the basic example copy pasted from the above manual. Below is the logs output of the tool.

python ./nodeset_compiler.py --types-array=UA_TYPES --existing ../../tools/schema/Opc.Ua.NodeSet2.PubSubMinimal.xml --xml ../../examples/nodeset/myNS.xml myNS
INFO:__main__:Preprocessing (existing) ../../tools/schema/Opc.Ua.NodeSet2.PubSubMinimal.xml
Traceback (most recent call last):
  File "./nodeset_compiler.py", line 123, in <module>
    ns.addNodeSet(xmlfile, True, typesArray=getTypesArray(nsCount))
  File "/home/alex/workspaces/open62541/tools/nodeset_compiler/nodeset.py", line 296, in addNodeSet
    node.replaceNamespaces(namespaceMapping)
  File "/home/alex/workspaces/open62541/tools/nodeset_compiler/nodes.py", line 184, in replaceNamespaces
    ref.referenceType.ns = nsMapping[ref.referenceType.ns]
AttributeError: 'unicode' object has no attribute 'ns'

## Background Information / Reproduction Steps
Follow the https://open62541.org/doc/current/nodeset_compiler.html:
1. Create myNS.xml from the provided code snippet
2. Try to run the provided command. No Opc.Ua.NodeSet2.xml found in the repo, so I replaced it with various Opc.Ua.NodeSet2.*.xml files, the result is the same for each file tried.
3. I am constantly receiving the same output every time.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [* ] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): none
 - [ *] Operating system: ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached N/A
 - [ ] Wireshark network dump attached N/A
 - [ *] Self-contained code example attached: yes
 - [ ] Critical issue:

simeonthefirst, 2020-12-11:
1. You can put multiple "--existing"  but i think you have to use the ns0  (--existing ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml)

2. Use  "git submodule update --init --recursive"  to clone the nodeset files, see: [https://open62541.org/doc/current/installing.html](url)

cmbahadir, 2021-01-02:
I was also facing with the same issue, and i have managed to compile it with `Opc.Ua.NodeSet2.Reduced.xml`

NoelGraf, 2021-04-16:
Were you able to fix the problem? I would close the problem for now. Please reopen this one if required.

closed 2021-04-16 by NoelGraf