open62541/open62541 #4058: Discovery Example

twolf-afk, 2020-12-09:
## Implementing a Local Discovery Server
I want to implement a local discovery server and have worked on the example in [open62541 Discovery Example](https://github.com/open62541/open62541/tree/master/examples/discovery).

`config->mdnsConfig.serverCapabilitiesSize = 1;`
This line is from the _server_lds.c_ file and the error message is, that `UA_ServerConfig` doesn't have a member `mdnsConfig`

`// periodic server register after 10 Minutes, delay first register for 500ms`
`UA_UInt64 callbackId;`
`UA_StatusCode retval = UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DISCOVERY_SERVER_ENDPOINT, 10 * 60 * 1000, 500, &callbackId);`
Those four lines are from _server_register.c_ where the server is registered at the LDS. The error message here is: `UA_Server_addPeriodicServerRegisterCallback` and `DISCOVERY_SERVER_ENDPOINT` are not defined.

So my question is, where can I find a working LDS example and how can I register my OPC UA Server?

It will help me a lot, if anyone has an idea on how to solve this !!

RakshanPremsagar, 2020-12-13:
If you are using Eclipse for development, make sure you use the open62541.c in the same folder as the src. 
Even i am beginning to explore discovery mechanism, i found the LDS to be wokŕking . So may be you could try it

twolf-afk, 2020-12-14:
Hey @RakshanPremsagar,

I am using MS Visual Studio 2019 for development.  I put the `open62541.c` and `open62541.h` in the src directory of my project and included the header file, but the error message is still the same.

RakshanPremsagar, 2020-12-14:
oh sorry, I am using linux, and I remember getting the same issue.. and it was fixed by adding the open62541.c file to the src folder. The LDS got working

RakshanPremsagar, 2020-12-30:
hey, have you managed to get it working?

cmbahadir, 2021-01-02:
Hello @twolf-afk,

> This line is from the _server_lds.c_ file and the error message is, that `UA_ServerConfig` doesn't have a member `mdnsConfig`

Well seems like you are missing build arguments for discovery service. Please make sure that you have set these arguments to 'ON' while building open62541.

`UA_ENABLE_DISCOVERY`                                                                                                                                                                                                                                             
`UA_ENABLE_DISCOVERY_MULTICAST`                                                                                                                                                                                                                                  
`UA_ENABLE_DISCOVERY_SEMAPHORE`

> Those four lines are from _server_register.c_ where the server is registered at the LDS. The error message here is: `UA_Server_addPeriodicServerRegisterCallback` and `DISCOVERY_SERVER_ENDPOINT` are not defined.

Actually, DISCOVERY_SERVER_ENDPOINT is defined in server_register.c [here](https://github.com/open62541/open62541/blob/78a6721bcbf932f74446d023370ab203cd48c093/examples/discovery/server_register.c#L17), it is not possible to say something about the error w/o seeing your example code. 

But again `UA_Server_addPeriodicServerRegisterCallback` depends on UA_ENABLE_DISCOVERY argument, so it should go away and you could manage to compile your example if you set the discovery service arguments above.

RakshanPremsagar, 2021-01-02:
Hello @cmbahadir ,
Is the UA_ENABLE_DISCOVERY_SEMAPHORE build option mandatory?

cmbahadir, 2021-01-02:
Hello @RakshanPremsagar 

No, as far as i can say its not required if the registration is not managed over semaphore files.

RakshanPremsagar, 2021-01-03:
Hello @cmbahadir ,
I have one query on Discovery mechanism.

I have a lds running(server_lds.c) and then I have another server(multicast.c) running on my system. The multicast server registers with the LDS and everything is fine. When i run the findServersOnNetwork code, I see both the servers.

However, when I close the multicast server, it should get deregistered from the LDS. 
But i get a log saying : Could not remove mDNS record for hostname Sample Multicast Server.

I get TXT and PTR entries on the  wireshark log as a result.
Could you tell me why does this happen? 
I have tried to debug by putting breakpoints on the log statement.. But when doing so, the multicast server does not get registered.
Could you Please tell me why does this happen?

cmbahadir, 2021-01-05:
Well, i guess a separate issue with required logs, traces and used version would be better for your problem.

I am not really familiar with discovery server architecture, so i can't answer your questions directly.

jackybek, 2021-07-19:
The reason for the missing structure member is you forgot to run 
git submodule update --init --recursive

o6-nbeier, 2026-10-20:
Closing for now.

closed 2026-10-20 by o6-nbeier