open62541/open62541 #4060: Travis is perhaps not usable in 2021

HolgerJeromin, 2020-12-09:
Just a heads up:

https://travis-ci.community/t/extremely-poor-official-communication-of-the-org-shutdown/10568/6

Not sure if this repo will be affacted.

P.S.: Do not forget https://github.com/open62541/open62541-www

ccvca, 2020-12-14:
Github-actions might be used instead, the following projects build open62541 and execute (my) tests using ctest:

Windows/Linux with Encryption, Docker container for to github container registry (tag by branch name), and handle pull requests and push (no errors due to missing secrets on pull requests)
https://github.com/umati/Sample-Server/tree/develop

And also with mdns:
https://github.com/seronet-project/SeRoNet-OPC-UA-Backend

To keep the ci-code clean, an extra CMake-Project which define all the CMake parameters/configurations using ExternalProject_Add:
https://github.com/umati/Sample-Server/blob/c1ddba8e172e7e6801d748fc797a8f61b1de0049/.github/CMakeLists.txt

ccvca, 2020-12-16:
Looks like OSS projects can be free (https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing):
> * We will be offering an allotment of OSS minutes that will be reviewed and allocated on a case by case basis. Should you want to apply for these credits please open a request with Travis CI support stating that you’d like to be considered for the OSS allotment. Please include:
>  * Your account name and VCS provider (like travis-ci.com/github/[your account name] ) How many credits (build minutes) you’d like to request (should your run out of credits again you can repeat the process to request more or discuss a renewable amount)
>  * Usage will be tracked under your account information so that you can better understand how many credits/minutes are being used

NoelGraf, 2021-04-16:
If I remember correctly, Travis is no longer supported. Github Actions will be used instead. Would then close the issue.

closed 2021-04-16 by NoelGraf

HolgerJeromin, 2021-04-16:
Closed by #4101