open62541/open62541 #4062: nodeset generation in CMakeLists

Sarutobie, 2020-12-10:
## Description
I am trying to use the DI nodeset according to [https://open62541.org/doc/current/nodeset_compiler.html](url) and the examples. I added the following lines to my CMakeLists.txt after the NS0 nodeset generation and before library building.

```
ua_generate_nodeset_and_datatypes(
        NAME "di"
        FILE_CSV "${PROJECT_SOURCE_DIR}/nodeset/DI/OpcUaDiModel.csv"
        FILE_BSD "${PROJECT_SOURCE_DIR}/nodeset/DI/Opc.Ua.Di.Types.bsd"
        NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
        FILE_NS "${PROJECT_SOURCE_DIR}/nodeset/DI/Opc.Ua.Di.NodeSet2.xml"
        DEPENDS "ns0"
        INTERNAL
    )
```

cmake then fails with the following error:

> Nodeset dependency ns0 needs to be generated before di.

## Background Information / Reproduction Steps

Used CMake options:

```BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu 20
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

simeonthefirst, 2020-12-11:
@Sarutobie  you don't need to set namespace0 manually as it is the default. (see https://github.com/open62541/open62541/blob/master/tools/cmake/macros_public.cmake, line 550)

Sarutobie, 2020-12-11:
That helped, thanks. What i still dont understand is if i still have to manually compile a nodeset using `nodeset_compiler.py`.
When i integrate DI nodeset in my CMakeLists.txt no additional source files are generated like it is stated in the documentation.
E.g. types_di_generated.c ...

ccvca, 2020-12-14:
This works for me:
https://github.com/umati/Sample-Server/blob/e79c062cce435b343d8e6fdd143507dcf28034ed/CMakeLists.txt#L20-L36
```cmake
ua_generate_nodeset_and_datatypes(
  NAME "di"
  FILE_NS "${open62541_NODESET_DIR}/DI/Opc.Ua.Di.NodeSet2.xml"
  FILE_CSV "${open62541_NODESET_DIR}/DI/OpcUaDiModel.csv"
  FILE_BSD "${open62541_NODESET_DIR}/DI/Opc.Ua.Di.Types.bsd"
  NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
  OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
)
# Add also heder files to trigger the generation!
add_library(ua_gen_di
  src_generated/namespace_di_generated.c
  src_generated/types_di_generated.h
  src_generated/di_nodeids.h
)
target_link_libraries(ua_gen_di PUBLIC open62541::open62541)
```

The target files (also the .h ones!) needs to be a dependency of your project to trigger the generation, so to use the above code also add this:
> target_link_libraries(YOUR_PROJECT PUBLIC ua_gen_di)

Sarutobie, 2020-12-16:
Thanks so far. This was missing. Looks like i have to dig in cmake a little bit, because this is not described in the documentation.
No DI-Nodeset works fine but if i try to add another nodeset, which then depends on DI the same way, it does not work.

maybe it is worth to mention, that i don´t have a project for this. I am adding this to CMakeLists of open62541 and building it.

ccvca, 2020-12-17:
Did you already have a look here: https://github.com/umati/Sample-Server/blob/e79c062cce435b343d8e6fdd143507dcf28034ed/CMakeLists.txt#L59-L74
This file is generating namespaces for: DI, IA (Industrial Automation), Machinery (using a hack to overcome missing bsd file), MT (Machine Tool)
```
              --------
              |- DI -|
              --------
                 /\
                /  \
               /    \
              /      \
             V        V
        --------  ---------------
        |- IA -|  |- Machinery -|
        --------  ---------------
            \            /
             \          /
              \        /
               \      /
                \    /
                 \  /
                  \/
               --------
               |- MT -|
               --------
```

Sarutobie, 2020-12-17:
Hi thank you for your answer. Of course i looked into your example.
And i am doing the same thing as i posted before.

I am not using the following line, as i am adding this to open62541 and not to my project.
> target_link_libraries(YOUR_PROJECT PUBLIC ua_gen_di)

I also tried adding IA as in your example and this also does not work.
Looks like i am missing something.

ccvca, 2020-12-17:
Can you please also post your `ua_generate_nodeset_and_datatypes` section. There seems to be an inconsitancy in the paths.

Sarutobie, 2020-12-17:
`ua_generate_nodeset_and_datatypes(
        NAME "di"
        FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml"
        FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/OpcUaDiModel.csv"
        FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/Opc.Ua.Di.Types.bsd"
        NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
        OUTPUT_DIR "${PROJECT_BINARY_DIR}/src_generated/open62541"
    )
    
    ua_generate_nodeset_and_datatypes(
	NAME "TEST"
	FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/TEST/test-v1.3.xml"
	FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/TEST/test-v1.3.csv"
	FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/TEST/test-v1.3.bsd"
    NAMESPACE_MAP "3:http://test.com/test/TEST/"
    OUTPUT_DIR "${PROJECT_BINARY_DIR}/src_generated/open62541"
    DEPENDS "di"

)`

Sarutobie, 2020-12-17:
Ok i found the problem.

The generated files are case sensitive.
So if the Nodeset is TEST, add_library also has to use TEST, not test.
correct would be:
```
add_library(ua_gen_TEST
  src_generated/open62541/namespace_TEST_generated.c
  src_generated/open62541/types_TEST_generated.h
  src_generated/open62541/TEST_nodeids.h
)
```

Sarutobie, 2020-12-22:
I had time again to work on this. 
Now i have nodeset DI and my own Nodeset in the CMakeLists.
cmake works fine
while make install i get the following error:

error: ‘Role_3’ undeclared (first use in this function)
 9027 | variablenode_ns_2_i_6178_variant_DataContents[0].criteriaType = (UA_Int32) Role_3;

As far as i see Role_3 is defined in OPC.UA.Types. Now i am not sure where the problem comes from. Is it my own nodeset or is an option missing in the building process.

types_genereted.c has 
`/* IdentityCriteriaType */
#define IdentityCriteriaType_members NULL`

ccvca, 2020-12-22:
Have you build open62541 with FULL namespace 0?

Sarutobie, 2020-12-22:
yes i have, see first post under used  CMake Options

ccvca, 2020-12-22:
Can you find the definition of "Role_3" in any header file? Is this header included in the source file by the nodeset compiler?

Sarutobie, 2020-12-23:
types_generated.h has the following. Other than that, no definition of role_3

```
/**
 * IdentityCriteriaType
 * ^^^^^^^^^^^^^^^^^^^^
 */
typedef enum {
    UA_IDENTITYCRITERIATYPE_USERNAME = 1,
    UA_IDENTITYCRITERIATYPE_THUMBPRINT = 2,
    UA_IDENTITYCRITERIATYPE_ROLE = 3,
    UA_IDENTITYCRITERIATYPE_GROUPID = 4,
    UA_IDENTITYCRITERIATYPE_ANONYMOUS = 5,
    UA_IDENTITYCRITERIATYPE_AUTHENTICATEDUSER = 6,
    __UA_IDENTITYCRITERIATYPE_FORCE32BIT = 0x7fffffff
} UA_IdentityCriteriaType;
UA_STATIC_ASSERT(sizeof(UA_IdentityCriteriaType) == sizeof(UA_Int32), enum_must_be_32bit);
```

ccvca, 2020-12-23:
Ok, do you known which part of the xml causes this role_3 assignment?

Sarutobie, 2020-12-23:
Yes there are two variables defined in the nodeset i am using.

ccvca, 2020-12-23:
Can you post the xml with the Variable where the role_3 is assigned and the type where role_3 is defined? Either there is an error in the xml or in the nodeset generator.
Do you generated your XML by hand or by a tool?

Sarutobie, 2021-02-09:
Thanks for the help so far. I figured out my problems and i can now use my nodeset.

closed 2021-02-09 by Sarutobie