open62541/open62541 #4068: error message when set up two dimensional array

fijo1015, 2020-12-11:
## Description

Hi, 

I am new with OPC UA. For one node i want to set up a two dimensional array. The code snippet is the following

> 	const unsigned int array_size = 2;
> 	UA_UInt32 arrayDims[1] = { array_size };
> 	
> 
> 	UA_VariableAttributes m_VariableAttr = UA_VariableAttributes_default;
> 	m_VariableAttr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
> 	m_VariableAttr.valueRank = UA_VALUERANK_TWO_DIMENSIONS;
> 	m_VariableAttr.arrayDimensions = (UA_UInt32 *)UA_Array_new(2, &UA_TYPES[UA_TYPES_DOUBLE]);
> 	m_VariableAttr.arrayDimensionsSize = 2;
> 	m_VariableAttr.arrayDimensions[0] = *arrayDims;
> 	m_VariableAttr.arrayDimensions[1] = *arrayDims;
> 	m_VariableAttr.displayName = UA_LOCALIZEDTEXT_CONST("en-US", "TestNode");
> 	m_VariableAttr.description = UA_LOCALIZEDTEXT_CONST("en-US", "");
> 	m_VariableAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
> 
> 	m_VariableAttr.value.arrayDimensions= (UA_UInt32 *)UA_Array_new(2, &UA_TYPES[UA_TYPES_DOUBLE]);
> 	m_VariableAttr.value.arrayDimensionsSize = 2;
> 	m_VariableAttr.value.arrayDimensions[0] = *arrayDims;
> 	m_VariableAttr.value.arrayDimensions[1] = *arrayDims;

> 	UA_Double values[4] = {0};
> 	UA_Variant_setArray(&m_VariableAttr.value, values, 4, &UA_TYPES[UA_TYPES_DOUBLE]);
 	
>
> 	UA_StatusCode retval = UA_Server_addVariableNode(server,
> 		UA_NODEID_NUMERIC(1, NULL),		// new node id
>                 UA_NODEID_NUMERIC(0, UA_NS0ID_ROOTFOLDER),
> 		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
> 		UA_QUALIFIEDNAME_CONST(1, "BrowseName TestArrayMultiDimensions"),
> 		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
> 		m_VariableAttr, NULL, &NodeId);	

But if I run this, I get the following error message. 

> WriteRequest returned status code BadTypeMismatch
> AddNodes: The value of ns=1;i=58192 is incompatible with the variable definition
> AddNodes: Type-checking the variable node ns=1;i=58192 failed with error code BadTypeMismatch

Instead of "UA_Double values[4] = {0}" I also tried "UA_Double values[2][2] = {0}". But the same error occurs. 

When I am leaving out  "UA_Double values[4] = {0}" and "UA_Variant_setArray(&m_VariableAttr.value, values, 4, &UA_TYPES[UA_TYPES_DOUBLE])" no error messages occur.

Can anybody tell ma what am I doing wrong? I really don´t know what´s my fault. 

 
I would be greatful for any help.

Best regards
jo

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [x] Other OPC UA SDKs used (client or server): UAExpert
 - [x] Operating system: Windows 10 Enterprise
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-12:
Update to the current master.
The log messages should be much more helpful.

fijo1015, 2020-12-14:
@jpfr Do you know, what I am doing wrong?

jpfr, 2020-12-31:
You set ArrayDimensions in the variable node.
But you also have to set the (same) arraydimensions in the value (variant).

NoelGraf, 2021-04-16:
I will close the issue for now. Reopen this one if required.

closed 2021-04-16 by NoelGraf