open62541/open62541 #4072: submodule dependencies

maldag, 2020-12-15:
Integrating this repository in our build system does not lock the gitmodules dependencies to certain hashes.

This can cause serious problems in this repository. 

Have you considered locking git submodules to hashes instead of branches?

It will break builds for older releases once the tracked branch progresses

ccvca, 2020-12-15:
Git submodules uses hashes by default, e.g. here in this implementation:
https://github.com/umati/Sample-Server/tree/main/deps

Make sure, that your .gitmodules does **not** define branches:
https://github.com/umati/Sample-Server/blob/e79c062cce435b343d8e6fdd143507dcf28034ed/.gitmodules

maldag, 2020-12-16:
I'm trying to point out, that releases tagged in this repository are in a stale state. 

We are fixing the hashes for our private fork.

NoelGraf, 2021-04-16:
I would close the issue. The "workaround" regarding the submodule is described in the documentation. Reopen this one if required.

closed 2021-04-16 by NoelGraf