open62541/open62541 #4074: Not able to set/increase MaxArrayLength

fijo1015, 2020-12-17:
## Description

Hi,
I want to increase the MaxArrayLength, which is a property of the array node. The UAExpert shows this:

[image]

Now I want to set the maxArrayLength value using the following code. 

> 	UA_VariableAttributes m_VariableAttr = UA_VariableAttributes_default;
> 	m_VariableAttr.displayName = UA_LOCALIZEDTEXT_CONST("en-US", "MaxArrayValue");
> 	m_VariableAttr.description = UA_LOCALIZEDTEXT_CONST("en-US", "");
> 	m_VariableAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
> 	m_VariableAttr.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_MAXARRAYLENGTH);
> 	UA_UInt32 maxvalue =  70000;
> 	UA_Variant_setScalarCopy(&m_VariableAttr.value, &maxvalue, &UA_TYPES[UA_TYPES_UINT32]),
> 
> 	UA_Server_addVariableNode(server,
> 		UA_NODEID_NUMERIC(1, NULL),
> 		UA_NODEID_NUMERIC(1, 54441),
> 		UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
> 		UA_QUALIFIEDNAME_CONST(1, "MaxArrayValue"),
> 		UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
> 		m_VariableAttr,
> 		NULL,
> 		NULL);

If I run this, I get the following error messages:

> WriteRequest returned status code BadTypeMismatch
> AddNodes: The value of ns=1;i=54442 is incompatible with the variable definition
> AddNodes: Type-checking the variable node ns=1;i=54442 failed with error code BadTypeMismatch

If I leave out the "setScalarCopy"-Line, then there is no Error Message anymore but the MaxArraySize Value has not been set. Can anybody tell me, what´s my fault?

Best regards
jo

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-31:
Which version of the sek do you use?

Later versions have much better error logs on the type checking.

The required type is MaxArray. But you try to set an Int32. Are the two compatible?

jpfr, 2020-12-31:
Also, try with the latest 1.2-rc1.
At least you should get better logs.

NoelGraf, 2021-04-16:
Have you tried the suggested solutions? Does the problem still exist?

NoelGraf, 2023-07-25:
Close the issue for now, reopen it if necessary.

closed 2023-07-25 by NoelGraf