open62541/open62541 #4075: Heap-use-after-free on session timeout during connection interruption

basyskom-jvoe, 2020-12-18:
If the connection is interrupted  (network cable pulled) when the session timeout occurs, there is a heap-use-after-free after the network cable is plugged back in:

```
==11469==ERROR: AddressSanitizer: heap-use-after-free on address 0x6060005112e0 at pc 0x7fe04a08aeee bp 0x7fe049b92d70 sp 0x7fe049b92d60

READ of size 8 at 0x6060005112e0 thread T49 (QThread)

    #0 0x7fe04a08aeed in UA_SecureChannel_processCompleteMessages /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:19688

    #1 0x7fe04a0dc31e in receiveServiceResponse /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:37951

    #2 0x7fe04a0dc638 in __UA_Client_Service /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:37985

    #3 0x7fe04a0e32bf in sendCloseSession /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:39205

    #4 0x7fe04a0e3717 in UA_Client_disconnect /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:39233

    #5 0x7fe04a0f28b8 in UA_Client_Subscriptions_processPublishResponse /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:41761

    #6 0x7fe04a0f30f4 in processPublishResponseAsync /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:41841

    #7 0x7fe04a0db40a in processAsyncResponse /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:37819

    #8 0x7fe04a0db946 in processServiceResponse /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:37858

    #9 0x7fe04a08a802 in processMessage /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:19656

    #10 0x7fe04a08ad6e in UA_SecureChannel_processCompleteMessages /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:19701

    #11 0x7fe04a0dc31e in receiveServiceResponse /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:37951

    #12 0x7fe04a0f43b2 in UA_Client_run_iterate /home/jvoe/Projects/qtopcua/src/3rdparty/open62541/open62541.c:42046

    #13 0x7fe04a17a15e in Open62541AsyncBackend::iterateClient() /home/jvoe/Projects/qtopcua/src/plugins/opcua/open62541/qopen62541backend.cpp:1032

    #14 0x7fe04a1a0460 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Open62541AsyncBackend::*)()>::call(void (Open62541AsyncBackend::*)(), Open62541AsyncBackend*, void**) (/usr/local/Qt-5.15.1/plugins/opcua/libopen62541_backend.so+0x184460)

    #15 0x7fe04a19e37d in void QtPrivate::FunctionPointer<void (Open62541AsyncBackend::*)()>::call<QtPrivate::List<>, void>(void (Open62541AsyncBackend::*)(), Open62541AsyncBackend*, void**) (/usr/local/Qt-5.15.1/plugins/opcua/libopen62541_backend.so+0x18237d)

    #16 0x7fe04a19881e in QtPrivate::QSlotObject<void (Open62541AsyncBackend::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (/usr/local/Qt-5.15.1/plugins/opcua/libopen62541_backend.so+0x17c81e)

    #17 0x7fe0818347bf  (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x2eb7bf)

    #18 0x7fe081839456 in QTimer::timeout(QTimer::QPrivateSignal) (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x2f0456)

    #19 0x7fe081839757 in QTimer::timerEvent(QTimerEvent*) (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x2f0757)

    #20 0x7fe08182b892 in QObject::event(QEvent*) (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x2e2892)

    #21 0x7fe0817f9bc0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x2b0bc0)

    #22 0x7fe081858058 in QTimerInfoList::activateTimers() (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x30f058)

    #23 0x7fe081858918  (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x30f918)

    #24 0x7fe07b8dc196 in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a196)

    #25 0x7fe07b8dc3ef  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a3ef)

    #26 0x7fe07b8dc49b in g_main_context_iteration (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4a49b)

    #27 0x7fe081858cab in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x30fcab)

    #28 0x7fe0817f8469 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0x2af469)

    #29 0x7fe0815fd36b in QThread::exec() (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0xb436b)

    #30 0x7fe0815fe727  (/usr/local/Qt-5.15.1/lib/libQt5Core.so.5+0xb5727)

    #31 0x7fe0813336b9 in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76b9)
[... 181 lines cut ...]
  Stack use after scope:   f8

  Global redzone:          f9

  Global init order:       f6

  Poisoned by user:        f7

  Container overflow:      fc

  Array cookie:            ac

  Intra object redzone:    bb

  ASan internal:           fe

==11469==ABORTING
```
The issue can be reproduced for every attempt.

## Checklist

Please provide the following information:

[X ] open62541 Version (release number or git tag): 1.0

jpfr, 2020-12-19:
Is this really for v1.0 as indicated?

basyskom-jvoe, 2020-12-21:
@jpfr Yes, this happens with the v1.0 amalgamation build

closed 2024-07-02 by basyskom-jvoe