open62541/open62541 #4077: Can't set ApplicationURI on client implementation

logari84, 2020-12-20:
Dear all,

I am implementing an OPC UA client. I have an issue with the setting of ApplicationURI. Perhaps I am missing something vital in the process, but here is how I try to set it up:

 ` client = UA_Client_new();
    clientConfig = UA_Client_getConfig(client);

    if (securityModeMap.find(securityMode) == securityModeMap.end())
    {
        stringstream ss;
        ss << "Wrong security mode: " << securityMode;
        VLOG(TAG, ERROR, ss.str());
        return false;
    }
    clientConfig->securityMode = securityModeMap[securityMode];

    if(!certificatePath.empty()) {
        clientConfig->clientDescription.applicationUri = UA_STRING_ALLOC("urn:opc.test.client");
        std::cout << "1 Application URI is: " << clientConfig->clientDescription.applicationUri.data << std::endl;
        UA_ByteString certificate = loadFile(certificatePath.c_str());
        UA_ByteString privateKey  = loadFile(keyPath.c_str());

        std::cout << "2 Application URI is: " << clientConfig->clientDescription.applicationUri.data << std::endl;
        std::cerr << "Called with cert!" << std::endl;
        UA_ClientConfig_setDefaultEncryption(clientConfig, certificate, privateKey,
                                             NULL, 0, NULL, 0);

        std::cout << "3 Application URI is: " << clientConfig->clientDescription.applicationUri.data << std::endl;
        std::cout << "Security mode: " << clientConfig->securityMode << std::endl;
        UA_ByteString_clear(&certificate);
        UA_ByteString_clear(&privateKey);
    } else {
        std::cerr << "Called default!" << std::endl;
        UA_ClientConfig_setDefault(clientConfig);
    }`

I included some prints to show where the variable changes.I set clientConfig->clientDescription.applicationUri, using UA_STRING_ALLOC (red 1 in the picture), but after I call " UA_ClientConfig_setDefaultEncryption(clientConfig, certificate, privateKey, NULL, 0, NULL, 0);" it changes to its original value (red 2 in the picture).

[image]

Any idea why is this happening? How am I able to set applicationURI on my client?

ccvca, 2020-12-21:
The function `UA_ClientConfig_setDefaultEncryption` sets all variables of the configuration to default values, so all custom configurations must be done after calling this function.

logari84, 2020-12-21:
Should have seen that coming! Thank you very much for the reply!

closed 2020-12-21 by logari84