open62541/open62541 #4081: There are some internet ressources/documentary about opc ua?

MichaelKlin, 2020-12-21:
## Description
Hallo i search some internet ressources/documentary(web, pdf) about opc ua. I found only from opcfoundation the documentary. It can be on English or Germany. Sorry for my English.
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] Other OPC UA SDKs used (client or server):

jpfr, 2020-12-22:
Start with Wikipedia.

MichaelKlin, 2020-12-22:
=) Thank you @jpfr . I mean more about architecutre/architecture captures. Because at library of my University i can not found e-book about opc ua and in Internet at the moment i am not founding well  ressources. I found some books at amazon, but i don not buy them

TonjaHeinemann, 2021-01-05:
Hi, my colleagues and I recently started a list of the resources we know: https://github.com/iswunistuttgart/awesome-opcua

MichaelKlin, 2021-01-08:
@TonjaHeinemann Thank you)

closed 2021-01-08 by MichaelKlin