open62541/open62541 #4086: Initialize string attribute on object, defined via XML Nodeset

alexanderkrause, 2020-12-22:
## Description
I try to initialize a string attribute on an object which I have defined in a XML:
Object:
```
<UAObjectType IsAbstract="true" NodeId="ns=1;i=1001"
                  BrowseName="1:DeviceType">
    <DisplayName>DeviceType</DisplayName>
    <References>
        <Reference ReferenceType="HasSubtype" IsForward="false">
            i=58
        </Reference>
    </References>
</UAObjectType>
```
Attribute with problems:
```
<UAVariable DataType="String" ParentNodeId="ns=1;i=1001"
            NodeId="ns=1;i=1012" BrowseName="1:Device-Name"
            UserAccessLevel="3" AccessLevel="3" >
    <DisplayName>Device-Name</DisplayName>
    <References>
        <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
        <Reference ReferenceType="HasModellingRule">i=78</Reference>
        <Reference ReferenceType="HasComponent" IsForward="false">
            ns=1;i=1001
        </Reference>
    </References>
</UAVariable>
```
It's initialization function:
```
UA_StatusCode StringAttributeInit(UA_Server* server, UA_NodeId* device, UA_Int16 nsIndex, char* attributeName, char* attributeValue)
{
    /* Find the NodeId of the id child variable */
    UA_RelativePathElement rpe;
    UA_RelativePathElement_init(&rpe);
    rpe.referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT);
    rpe.isInverse = false;
    rpe.includeSubtypes = false;
    rpe.targetName = UA_QUALIFIEDNAME(nsIndex, attributeName);

    UA_BrowsePath bp;
    UA_BrowsePath_init(&bp);
    bp.startingNode = *device;
    bp.relativePath.elementsSize = 1;
    bp.relativePath.elements = &rpe;

    UA_BrowsePathResult bpr = UA_Server_translateBrowsePathToNodeIds(server, &bp);
    if(bpr.statusCode != UA_STATUSCODE_GOOD || bpr.targetsSize < 1)
    {
        return bpr.statusCode;
    }
    
    printf("String: %s\n", attributeValue);

    /* Set the status value */
    UA_Variant value;
    UA_String st = UA_String_fromChars("Test");
    UA_Variant_setArrayCopy(&value, st.data, st.length, &UA_TYPES[UA_TYPES_STRING]);

    UA_Server_writeValue(server, bpr.targets[0].targetId.nodeId, value);      //<-- here it does not work
    UA_BrowsePathResult_clear(&bpr);

    return UA_STATUSCODE_GOOD;
}
```

Attribute without problems: I can initialize the attribute with any uint32
```
<UAVariable DataType="UInt32" ParentNodeId="ns=1;i=1001"
            NodeId="ns=1;i=1011" BrowseName="1:Device-ID"
            UserAccessLevel="3" AccessLevel="3">
    <DisplayName>Device-ID</DisplayName>
    <References>
        <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
        <Reference ReferenceType="HasModellingRule">i=78</Reference>
        <Reference ReferenceType="HasComponent" IsForward="false">
            ns=1;i=1001
        </Reference>
    </References>
</UAVariable>
```

It's initialization function:
```
UA_StatusCode UInt32AttributeInit(UA_Server* server, UA_NodeId* device, UA_Int16 nsIndex, char* attributeName, UA_UInt32 attributeValue)
{
    /* Find the NodeId of the id child variable */
    UA_RelativePathElement rpe;
    UA_RelativePathElement_init(&rpe);
    rpe.referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT);
    rpe.isInverse = false;
    rpe.includeSubtypes = false;
    rpe.targetName = UA_QUALIFIEDNAME(nsIndex, attributeName);

    UA_BrowsePath bp;
    UA_BrowsePath_init(&bp);
    bp.startingNode = *device;
    bp.relativePath.elementsSize = 1;
    bp.relativePath.elements = &rpe;

    UA_BrowsePathResult bpr = UA_Server_translateBrowsePathToNodeIds(server, &bp);
    if(bpr.statusCode != UA_STATUSCODE_GOOD || bpr.targetsSize < 1)
    {
        return bpr.statusCode;
    }
        
    /* Set the status value */
    UA_Variant value;
    UA_Variant_setScalar(&value, &attributeValue, &UA_TYPES[UA_TYPES_UINT32]);
    UA_Server_writeValue(server, bpr.targets[0].targetId.nodeId, value);
    UA_BrowsePathResult_clear(&bpr);

    return UA_STATUSCODE_GOOD;
}
```
## Background Information / Reproduction Steps
[image]
As you can see the value I try to write does not appear in the object. I already tried `UA_Variant_setArray` command with different parameters ("Test" as parameter, a pointer to a predefined char array, different array lengths, ...). There is also the UA_Variant_setScalar() command, but there the BadOutOfMemory Error occurs. I also tried to change the type fo UA_TYPES_SBYTE, BYTE, ... with similar results (often BadTypeMismatch).

Console Log: https://paste2.org/M2XIeIgL

Used CMake options:

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): git master branch
 - [x] Other OPC UA SDKs used (client or server): UA Expert
 - [x] Operating system: Windows 10, but I build and execute with WSL 2 Ubuntu
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Level 200 https://paste2.org/M2XIeIgL, Level 100 results in a log which does not fit on my console and also throws a Segmentation fault 
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-23:
Don’t use  UA_Variant_setArrayCopy in the first example.
You are setting an array of many strings.

alexanderkrause, 2020-12-23:
> 
> 
> Don’t use UA_Variant_setArrayCopy in the first example.
> You are setting an array of many strings.

Thank you for your reply. I changed to UA_Variant_setScalarCopy but get BadTypeMismatch everytime. I tried to pass a UA_String, char*, and pointers to both. For the types as Argument for UA_Variant_setScalarCopy I use UA_TYPES_STRING. Which Type do I have to use (e.g. on char array)? Or what would be the correct "string" argument for  UA_TYPES_STRING?

closed 2026-10-20 by o6-nbeier