open62541/open62541 #4094: Library build with version tag v1.0.4  is not compatible to recent versions (i.e. tagged v1.0.3)

jrcoding, 2020-12-23:
## Description

open62541 library **version  tag v1.0.4  is not compatible to recent versions (i.e. tagged v1.0.3)** anymore.

## Background Information / Reproduction Steps

Applications build with open62541 library version tagged v1.0.3 could not be build with library version tagged v.1.0.4 anymore without changing the sources of the application. 

Seems to me that it's a result of a   _**Merge branch '1.1' into 1.0**_   (5481c42cb79e) ?!

**Is this by chance or intentionally ?**

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): **v1.0.4**
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: **Debian Linux** 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-12-24:
Grr. git played a trick here.
Use the new release v1.0.5 https://github.com/open62541/open62541/releases/tag/v1.0.5

closed 2020-12-24 by jpfr

jrcoding, 2020-12-25:
sorry, branch 1.0 still doesn't compile....
seems that the last commit is still a commit (06a3b0eaaf65a3c2c82c66d7913f531642dca13e) for branch 1.1