open62541/open62541 #4099: Error executing compiled server file

SoulpleQreun, 2020-12-30:
## Description

There was an error in the last step of building the server  

below：

[2020-12-30 13:10:00.403 (UTC+0800)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2020-12-30 13:10:00.404 (UTC+0800)] info/server	AccessControl: Anonymous login is enabled [2020-12-30 13:10:00.404 (UTC+0800)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-12-30 13:10:00.404 (UTC+0800)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-12-30 13:10:00.404 (UTC+0800)] error/server	Could not add the example nodeset. 

Is there a problem with my generation node  about information model 

How can I get through this problem？

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

MichaelKlin, 2020-12-30:
can you maybe more explain? or show some code

SoulpleQreun, 2020-12-31:
Thank you for your reply. I have solved this problem. I rebuild the information model，and the server.c file was also modified. 
Maybe it's a mistake in the server.c file

> can you maybe more explain? or show some code

NoelGraf, 2021-04-16:
Then I would close the issue.

closed 2021-04-16 by NoelGraf