open62541/open62541 #4100: Observation on the LDS log

RakshanPremsagar, 2020-12-30:
Hi there,
I have the server_lds.c running from the examples section, after which i run the multicast server.
Now I see advertisements with query and responses on Wireshark.
If I end the multicast server operation, I can see that on the log(Multicast server), it says remove record for :Sample Multicast Server-<hostname>._opcua-tcp._tcp.local. 

But on the LDS log, I see contradicting response like: Could not remove mDNS record for hostname Sample Multicast Server.

As a result, I see some advertisements even after multicast server has ended its operation.
Why does this happen? Am i doing anything wrong here?
Please let me know. 
-->

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: LINUX
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

cmbahadir, 2021-01-06:
Hello @RakshanPremsagar ,

Is this a duplicate of #4108 ?

RakshanPremsagar, 2021-01-06:
Yes, I will probably delete this one

closed 2021-01-06 by RakshanPremsagar