open62541/open62541 #4106: set Priority to the publisher messages

rass92, 2021-01-05:
Dear Community,

I am using the RT publisher and subscriber connection, and I need to edit the prioritization of the traffic that is been sent by the publisher. I am using the pubsub_TSN_loopback.c and pubsub_TSN_publisher.c.

Where can I edit the priority of the messages sent by the publisher? I understand that have to done at the socket level with the function "setsockopt", but I dont know where to do it. Does somebody have any hint

Thank you in advance

johndzilva, 2021-01-06:
You can assign the priority the publisher packets here,
https://github.com/open62541/open62541/blob/beeafc02e9f98c52f0dee2364620f5c3527da5db/examples/pubsub_realtime/pubsub_TSN_publisher.c#L534

After doing that you need to map the priority using qdisc command,
https://github.com/open62541/open62541/blob/beeafc02e9f98c52f0dee2364620f5c3527da5db/examples/pubsub_realtime/README_pubsub_TSN.txt#L37

rass92, 2021-01-06:
Thank you very much Johndzilva !! that is exactly what I need

rass92, 2021-01-07:
Johndzilva I came with another question this Priority can be put use if the config->pubsubTransportLayers[1]=UA_PubSubTransportLayerUDPMP();

johndzilva, 2021-01-08:
@rass92 no it is not possible to do with UA_PubSubTransportLayerUDPMP();.
Linux socket option SO_PRIORITY is used to set the socket priority.

rass92, 2021-01-10:
But do you know in which file is the socket of the Pub/Sub communication created so I can set the SO_PRIORITY?

johndzilva, 2021-01-14:
@rass92 
https://github.com/open62541/open62541/blob/e5eba7bd591642281717be9edbcd0b6d2ba0fe0b/plugins/ua_pubsub_ethernet.c#L324

rass92, 2021-03-09:
Dear johndzilva

thank you for your answer, another way give a priority to the package is with iptables.

The command is like this

```
iptables -t mangle -A POSTROUTING -p udp --dport 4010 -j CLASSIFY --set-class 0:4
```
Where the outgoing UDP packages that have destination port 4010 will have a priority 4.

closed 2021-03-09 by rass92