open62541/open62541 #4108: Local Discover Server - Log observation ( Un-Registering not successful)?

RakshanPremsagar, 2021-01-06:
Hi there,
I have the server_lds.c running with mDNS extension from the examples/Discovery section, I see the MDNS packets on the wireshark tool.
Next, I run the server_multicast.c. The mutlicast server gets registered to the LDS. I see MDNS packets corresponding to the multicast server as well.

Theoretically, if I stop the multicast server(gracefully with ctrl+c), It should unregister from the LDS. 
[image]
[image]
[image]

But I observe a SRV packet with TTL =0 Indicating that the service is no more.

But on the LDS log, I see "could not remove record for sample multicast server"
As a result, I see TXT packets on the wireshark tool.
And if i run findServOnNetwork code, I get incorrect results.

Please let me know what is going wrong here?

 - [x ] open62541 Version (release number or git tag):v1.2-rc1

 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

RakshanPremsagar, 2021-01-06:
Build option:cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON..

cmbahadir, 2021-01-06:
Seems like this is caused by the same root cause as #3789 .
So, adding a callback to `server_multicast` example solved the problem on my side.

```
UA_UInt64 callbackId;
retval = UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, endpointUrl,
                                                         10 * 60 * 1000, 500, &callbackId);
```
and remove after receiving SIGINT, removed the warning.
`UA_Server_removeCallback(server, callbackId);`

RakshanPremsagar, 2021-01-06:
Hello @cmbahadir , I tried doing the same. No success. Infact, I get the same log messages when I de-register the (server_register.c) file from LDS as well.
Is it any version issue that I am facing? I am not getting the clarity here

cmbahadir, 2021-01-06:
Hmm, yes it is already applied on `server_register.c`.

Have you add `UA_Server_removeCallback(server, callbackId);` before `UA_Server_run_shutdown(server);`, since shutdown flow in multicast server is a bit different?

RakshanPremsagar, 2021-01-06:
Yes I did that as well now. Still no success :(

RakshanPremsagar, 2021-01-06:
by any chance, is this getting replicated in your case? @cmbahadir

cmbahadir, 2021-01-06:
Well, yes warning still comes sorry for that and there are still unwanted characters in the end of the `serverName` in `*hashentry`, seems like callback doesn't solve it.

RakshanPremsagar, 2021-01-06:
Hi @cmbahadir ,

So when i run the findServers.c code, I get correct response(as though it has deregistered)
But findServersOnNet does not give me correct response. 
Could you check the same?

Also if possible, could you please share the modified lines of code here? I am working with the amalgamated file..

RakshanPremsagar, 2021-01-16:
Hello @cmbahadir ,
When you run the findServonNetwork code, does it show inconsistent readings? After the changes you suggested?

Regards,
Rakshan