open62541/open62541 #4117: How to parse ExtensionObject from InputArgument of method callback function

hhashoww, 2021-01-12:
## Description
I'm the newbie of open62541

I'm trying to find a way to parse the input argument with ExtensionObject type on my sample OPC UA server

This is the input argument from Wireshark
[wireshark snapshot](https://imgur.com/ABmaejT)

I hope I can parse the Bytestring then get the content string - `GG`

Below is my code:
```
static UA_StatusCode __add_test_method_cb(UA_Server *server, const UA_NodeId *sessionId, void *sessionHandle,
                                                 const UA_NodeId *methodId, void *methodContext,
                                                 const UA_NodeId *objectId, void *objectContext, size_t inputSize,
                                                 const UA_Variant *input, size_t outputSize, UA_Variant *output) {
    ...
    ...
    UA_ExtensionObject *eo = (UA_ExtensionObject *)input[0].data;
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoding is %d", eo->encoding); // is UA_EXTENSIONOBJECT_ENCODED_BYTESTRING 
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoded bytestring length is %d", (int)eo->content.encoded.body.length); // is 6
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoded bytestring length is %d", (int)eo->content.encoded.body.length); // is 2
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoded bytestring is %*.s", (int)eo->content.encoded.body.length, eo->content.encoded.body.data); // ???
}
```
I always get empty string on the console

I expect to get the input string from UaExpert

Could someone can help me?

Thanks

Used CMake options:

Compiled open62541 with

```bash
cmake -DCMAKE_BUILD_TYPE=Release -DUA_ENABLE_AMALGAMATION=ON UA_NAMESPACE_ZERO=FULL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

hhashoww, 2021-01-13:
I try to use `UA_ByteString_toBase64()` to translate the UA_ByteString body in the UA_ExtensionObject
```
typedef struct {
    UA_ExtensionObjectEncoding encoding;
    union {
        struct {
            UA_NodeId typeId;   /* The nodeid of the datatype */
            UA_ByteString body; /* The bytestring of the encoded data */
        } encoded;
        struct {
            const UA_DataType *type;
            void *data;
        } decoded;
    } content;
} UA_ExtensionObject;
```

```
  UA_String ns;
  UA_String_init(&ns); /* _init zeroes out the entire memory of the datatype */
  UA_StatusCode ret = UA_ByteString_toBase64(&eo->content.encoded.body, &ns);
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "ret is %d", ret); // UA_STATUSCODE_GOOD 
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "The NS length is %d", (int)ns.length); // is 8
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "The NS is %*.s", (int)ns.length, ns.data); // still empty
```

hhashoww, 2021-01-13:
I found I use the wrong way to parse the input argument
the `typekind` of input variant is 21 (UA_TYPES_EXTENSIONOBJECT )

So I use the way to parse the variant
```
  UA_ExtensionObject eo;
  UA_ExtensionObject_init(&eo);
  UA_ExtensionObject_setValue(&eo, input[0].data, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoding is %d", eo.encoding); // is 3 (UA_EXTENSIONOBJECT_DECODED)
```

So I have to use decoded to explain the content in ExtensionObject

But I still have no idea how to go to the next step

I will keep solving this question and left the process

Hope it can help someone oneday

the-om, 2021-01-14:
Your strings only appear empty because your printf syntax is wrong, change `%*.s` to `%.*s`

hhashoww, 2021-01-14:
> Your strings only appear empty because your printf syntax is wrong, change `%*.s` to `%.*s`

Oops! I did stupid things !!!
@the-om  I am grateful for your help

But after fixing the print syntax to `%.*s` 
I still cannot parse the input argument

I think I need to find an example to parse the input while the DataType is not a Builtin Type
Keep trying~

the-om, 2021-01-14:
It is not clear what kind of data you are entering in UA_Expert. If you are passing a string, than the type of the input argument should be UA_String.
If you are passing some kind of custom data type, then you should make a type definition for that type and add it to the customDataTypes member in the server configuration (See examples/custom_datatype).

hhashoww, 2021-01-15:
> I found I use the wrong way to parse the input argument
> the `typekind` of input variant is 21 (UA_TYPES_EXTENSIONOBJECT )
> 
> So I use the way to parse the variant
> 
> ```
>   UA_ExtensionObject eo;
>   UA_ExtensionObject_init(&eo);
>   UA_ExtensionObject_setValue(&eo, input[0].data, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);
>   UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoding is %d", eo.encoding); // is 3 (UA_EXTENSIONOBJECT_DECODED)
> ```
> 
> So I have to use decoded to explain the content in ExtensionObject
> 
> But I still have no idea how to go to the next step
> 
> I will keep trying to solve my question and record the process
> 
> Hope it can help someone one day :)

It's wrong !!!!
This `encoding` value is set via `UA_ExtensionObject_setValue()`

I find a description of `Variant`
```
If the data type is unknown to the receiver, the variant contains the original ExtensionObject in binary or XML encoding.
```
And a description of `ExtensionObject`
```
If the received data type is unknown, the encoded string and target NodeId is stored instead of the decoded value.
```

I did a memory dump for the ExtensionObject's encoded part
```
0x02    0x00    0x00    0x00    0x00    0x00    0x00    0x00
0x94    0x13
```
It can be mapped to a reasonable UA_NODEID `ns=2;i=5012`
And it is a `HasEncoding` reference which is a **Binary Encoding**

It perfect match the description in the document above
So I think the next step is how to decode the **Binary Encoding** data in the encoded body

This is a complete memory dump of input->data with memsize(48 bytes) (which is described in the input variant data structure)
```
  0x01    0x00    0x00    0x00    0x00    0x00    0x00    0x00
  0x02    0x00    0x00    0x00    0x00    0x00    0x00    0x00
  0x94    0x13    0x00    0x00    0x00    0x00    0x00    0x00
  0x00    0x00    0x00    0x00    0x00    0x00    0x00    0x00
  0x06    0x00    0x00    0x00    0x00    0x00    0x00    0x00
  0x00    0x30    0xbc    0x01    0x00    0x00    0x00    0x00
```

hhashoww, 2021-01-15:
> It is not clear what kind of data you are entering in UA_Expert. If you are passing a string, than the type of the input argument should be UA_String.
> If you are passing some kind of custom data [...]

@the-om thanks for your reply
The DataType is a DataType that has `HasEncoding` reference type,
So I think I have to find a way to decode the encoded body,
but I remember the decode binary function is not exposed for the single file mode (open62541.c/open62541.h)
is it correct???

thanks

hhashoww, 2021-01-15:
I made it !!! 

I expose the function to `open62541.h`
```
UA_StatusCode
UA_decodeBinary(const UA_ByteString *src, size_t *offset, void *dst,
                const UA_DataType *type, const UA_DataTypeArray *customTypes);
```

And this is my code
```
  UA_ExtensionObject *eo = (UA_ExtensionObject *)input[0].data;
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "eo encoded numeric is %d", eo->content.encoded.typeId.identifier.numeric); // 5012, which is a Object node has binary encoding type

  UA_String ns;
  UA_String_init(&ns); /* _init zeroes out the entire memory of the datatype */
  size_t offset = 0;
  UA_decodeBinary(&eo->content.encoded.body, &offset, &ns, &UA_TYPES[UA_TYPES_STRING], NULL);
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "The NS length is %d", (int)ns.length); // is 2
  UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "The NS is %.*s", (int)ns.length, ns.data); // GG, I made it !!!
```

The next step is how to know the type of content of the encoded UA_Bytestring
I hard code it with `UA_String` now

jpfr, 2021-01-17:
Where the does the datatype ns=2;i=5012 come from?
If you configure this datatype in the server (see the example for custom datatypes), then it will be decoded automatically.
You should never have to use internal (private) APIs for normal operations.

hhashoww, 2021-01-18:
> Where the does the datatype ns=2;i=5012 come from?
> If you configure this datatype in the server (see the example for custom datatypes), then it will be decoded automatically.
> You should never have to use internal (private) APIs for normal operations.

@jpfr I confused that also, I just wrap a string to another DataType as my method's input type
[I Put the view from UaExpert here](https://imgur.com/a/HIqNs51)

## This is the definition of 5012
```
/* Default Binary - ns=2;i=5012 */

static UA_StatusCode function_my_test_model_addon_66_begin(UA_Server *server, UA_UInt16* ns) {
	UA_StatusCode retVal = UA_STATUSCODE_GOOD;
	UA_ObjectAttributes attr = UA_ObjectAttributes_default;
	attr.displayName = UA_LOCALIZEDTEXT("", "Default Binary");
	retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_OBJECT,
	UA_NODEID_NUMERIC(ns[1], 5012LU),
	UA_NODEID_NUMERIC(ns[0], 0LU),
	UA_NODEID_NUMERIC(ns[0], 0LU),
	UA_QUALIFIEDNAME(ns[0], "Default Binary"),
	UA_NODEID_NUMERIC(ns[0], 76LU),
	(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_OBJECTATTRIBUTES],NULL, NULL);
	retVal |= UA_Server_addReference(server, UA_NODEID_NUMERIC(ns[1], 5012LU), UA_NODEID_NUMERIC(ns[0], 38LU), UA_EXPANDEDNODEID_NUMERIC(ns[1], 3013LU), false);
	return retVal;
}

static UA_StatusCode function_my_test_model_addon_66_finish(UA_Server *server, UA_UInt16* ns) {
	return UA_Server_addNode_finish(server,
	UA_NODEID_NUMERIC(ns[1], 5012LU)
);
}
```

## This is my input argument of the test method
```
static UA_StatusCode function_my_test_model_addon_298_begin(UA_Server *server, UA_UInt16* ns) {
	UA_StatusCode retVal = UA_STATUSCODE_GOOD;
	UA_VariableAttributes attr = UA_VariableAttributes_default;
	attr.minimumSamplingInterval = 0.000000;
	attr.userAccessLevel = 1;
	attr.accessLevel = 1;
	attr.valueRank = 1;
	attr.arrayDimensionsSize = 1;
	UA_UInt32 arrayDimensions[1];
	arrayDimensions[0] = 1;
	attr.arrayDimensions = &arrayDimensions[0];
	attr.dataType = UA_NODEID_NUMERIC(ns[0], 296LU);
	UA_Argument variablenode_ns_1_i_6076_variant_DataContents[1];

	UA_init(&variablenode_ns_1_i_6076_variant_DataContents[0], &UA_TYPES[UA_TYPES_ARGUMENT]);
	variablenode_ns_1_i_6076_variant_DataContents[0].name = UA_STRING("myTestInputName");
	variablenode_ns_1_i_6076_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[1], 3013LU);
	variablenode_ns_1_i_6076_variant_DataContents[0].valueRank = (UA_Int32) -1;
	UA_Variant_setArray(&attr.value, &variablenode_ns_1_i_6076_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_ARGUMENT]);
	attr.displayName = UA_LOCALIZEDTEXT("", "InputArguments");
	retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
	UA_NODEID_NUMERIC(ns[1], 6076LU),
	UA_NODEID_NUMERIC(ns[1], 7022LU),
	UA_NODEID_NUMERIC(ns[0], 46LU),
	UA_QUALIFIEDNAME(ns[0], "InputArguments"),
	UA_NODEID_NUMERIC(ns[0], 68LU),
	(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLEATTRIBUTES],NULL, NULL);

	return retVal;
}

static UA_StatusCode function_my_test_model_addon_298_finish(UA_Server *server, UA_UInt16* ns) {
	return UA_Server_addNode_finish(server,
	UA_NODEID_NUMERIC(ns[1], 6076LU)
);
}
```

hhashoww, 2021-01-19:
did I misunderstand the definition of "Customer DataType"?
If the DataType is not a built-in type, it called "Customer DataType"?
```
variablenode_ns_1_i_6076_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[1], 3013LU);
```

So I have to define a data structure like the example in "examples/custom_datatype" to decode the Bytestring?

hhashoww, 2021-01-19:
I must say I'm completely wrong.

I use customer data type really, and I'm trying to follow the [XML Nodeset Compiler](https://open62541.org/doc/1.1/nodeset_compiler.html#) to generate the code

I have to provide `.csv` and `.bsd` files to generate the customer datatype definitions

hhashoww, 2021-01-20:
Now I put the following files under /deps/ua-nodeset/Schema/ folder (which is generated by UAModeler)
* myTest.csv
* myTest.bsd
* myTest.xml

Then I want to generate the code automatically with the tutorial, 
so I append the following commands into the top **CMakeLists.txt** (**or should I generate a sub-folder for the following steps?** like `example/myTest/`)
```
#########################
# Generate source files #
#########################

...
...
# Header containing defines for all NodeIds 
ua_generate_nodeid_header(
    NAME "nodeids"
    ID_PREFIX "NS0"
    TARGET_SUFFIX "ids-ns0"
    FILE_CSV "${UA_FILE_NODEIDS}"
)

################ Start Here ########################
# Generate types and namespace for myTest
ua_generate_nodeset_and_datatypes(
    INTERNAL
    NAME "myTest"
    FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/myTest.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/myTest.bsd"
    IMPORT_BSD "UA_TYPES#${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.Types.bsd"
    NAMESPACE_MAP "2:http://opcfoundation.org/myTest/"
    OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml"
)
)
################ End Here ########################
...
...
```

But I only get `types` files
* types_myTest_generated.c
* types_myTest_generated.h
* types_myTest_generated_encoding_binary.h
* types_myTest_generated_handling.h

I expect I can get `namespace` series files like the tutorial?

hhashoww, 2021-01-20:
Hi all & @Pro , I found a small issue about "ua_generate_nodeset_and_datatypes" API
The `FILE_NS` can accept multiple values as the input
```
FILE_NS         Path to the NodeSet2.xml file. Multiple values can be passed. These nodesets will be combined into one output.
```

But the argument set function still consider as the single value argument
```
set(oneValueArgs NAME FILE_NS FILE_CSV FILE_BSD OUTPUT_DIR TARGET_PREFIX BLACKLIST)
```

It should be corrected at `macros_public.cmake`
```
-   set(oneValueArgs NAME FILE_NS FILE_CSV FILE_BSD OUTPUT_DIR TARGET_PREFIX BLACKLIST)
-   set(multiValueArgs DEPENDS IMPORT_BSD NAMESPACE_MAP)
+   set(oneValueArgs NAME  FILE_CSV FILE_BSD OUTPUT_DIR TARGET_PREFIX BLACKLIST)
+   set(multiValueArgs DEPENDS FILE_NS IMPORT_BSD NAMESPACE_MAP)
```

hhashoww, 2021-01-21:
I finally figured out all the thing and what I should do

1. I have two nodesets `nodesetA` & `nodesetB`, `nodesetB` is based on `nodesetA`
2. There are lots of custom define **DataType** in the two nodesets
3. I append commands into the top **CMakeLists.txt**
```
# generate nodesetA (only for datatype, namespace_nodesetA_generated.c/.h is no use, maybe I should use `ua_generate_datatypes`  here)
 ua_generate_nodeset_and_datatypes(
    INTERNAL
    NAME "nodesetA"
    FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.bsd"
    NAMESPACE_MAP "2:http://mywebsite/UA/nodesetA"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.xml"
)
 # generate nodesetB namespace which is using nodesetA
 ua_generate_nodeset_and_datatypes(
    INTERNAL
    NAME "nodesetB"
    FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetB.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetB.bsd"
    NAMESPACE_MAP "3:http://mywebsite/UA/nodesetB"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetB.xml" "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.xml"
)
```
4. After that, I can get the following files:
* nodesetA_nodeids.h
* nodesetB_nodeids.h
* namespace_nodesetA_generated.c
* namespace_nodesetA_generated.h
* namespace_nodesetB_generated.c
* namespace_nodesetB_generated.h
* types_nodesetA_generated.c
* types_nodesetA_generated.h
* types_nodesetA_generated_encoding_binary.h
* types_nodesetA_generated_handling.h
* types_nodesetB_generated.c
* types_nodesetB_generated.h
* types_nodesetB_generated_encoding_binary.h
* types_nodesetB_generated_handling.h

## But there are some questions here
1. It seems an issue here, if I use **DEPENDS** to refer nodesetA in nodesetB
the namespace index is not correct at the dataType
It should be `UA_NODEID_NUMERIC(ns[2], 3013LU);`
So I have to use the **FILE_NS** to combine two nodesets into one output.
### CMakeLists.txt
```
# generate nodesetA (only for datatype, namespace_nodesetA_generated.c/.h is no use, maybe I should use `ua_generate_datatypes`  here)
 ua_generate_nodeset_and_datatypes(
    INTERNAL
    NAME "nodesetA"
    FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.bsd"
    NAMESPACE_MAP "2:http://mywebsite/UA/nodesetA"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetA.xml"
)
 # generate nodesetB namespace which is using nodesetA
 ua_generate_nodeset_and_datatypes(
    INTERNAL
    NAME "nodesetB"
    FILE_CSV "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetB.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetB.bsd"
    NAMESPACE_MAP "3:http://mywebsite/UA/nodesetB"
    FILE_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/nodesetB.xml"
    DEPENDS "nodesetA"
)
```
### namespace_nodesetB_generated.c
```
/* InputArguments - ns=2;i=6076 */

static UA_StatusCode function_namespace_nodesetB_generated_64_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
attr.valueRank = 1;
attr.arrayDimensionsSize = 1;
UA_UInt32 arrayDimensions[1];
arrayDimensions[0] = 1;
attr.arrayDimensions = &arrayDimensions[0];
attr.dataType = UA_NODEID_NUMERIC(ns[0], 296LU);
UA_Argument variablenode_ns_2_i_6076_variant_DataContents[1];

UA_init(&variablenode_ns_2_i_6076_variant_DataContents[0], &UA_TYPES[UA_TYPES_ARGUMENT]);
variablenode_ns_2_i_6076_variant_DataContents[0].name = UA_STRING("myName");
variablenode_ns_2_i_6076_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[1], 3013LU);
variablenode_ns_2_i_6076_variant_DataContents[0].valueRank = (UA_Int32) -1;
UA_Variant_setArray(&attr.value, &variablenode_ns_2_i_6076_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_ARGUMENT]);
attr.displayName = UA_LOCALIZEDTEXT("", "InputArguments");
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
UA_NODEID_NUMERIC(ns[2], 6076LU),
UA_NODEID_NUMERIC(ns[2], 7022LU),
UA_NODEID_NUMERIC(ns[0], 46LU),
UA_QUALIFIEDNAME(ns[0], "InputArguments"),
UA_NODEID_NUMERIC(ns[0], 68LU),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLEATTRIBUTES],NULL, NULL);

return retVal;
}
```

2. The **FILE_NS** should support multiple value input
3. The `customTypesSize` argument of `UA_decodeBinary()` should be removed from generated types file
4. The `src` argument of `UA_calcSizeBinary()` should be casted
```
In file included from open62541_method_cb.c:5:
types_nodesetB_generated_encoding_binary.h: In function 'UA_MyDataType_calcSizeBinary':
types_nodesetB_generated_encoding_binary.h:173:30: warning: passing argument 1 of 'UA_calcSizeBinary' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  173 |     return UA_calcSizeBinary(src, &UA_TYPES_NODESETB_ADDON[UA_TYPES_NODESETB_ADDON_MYDATATYPE]);
      |                              ^~~
In file included from open62541_method_cb.c:2:
namespace_nodesetB_generated.h:30:25: note: expected 'void *' but argument is of type 'const UA_MyDataType *' {aka 'const struct <anonymous> *'}
   30 | UA_calcSizeBinary(void *p, const UA_DataType *type);
      |                   ~~~~~~^
```

closed 2021-01-21 by hhashoww