open62541/open62541 #4119: Multithreading regression

rexsum, 2021-01-13:
## Description
Assert failure locking/unlocking mutex

## Background Information / Reproduction Steps

When shutting down a PubSub server an assert failure occurs. This is easy to reproduce, simply build the library with UA_MULTITHREADING=199, then compile and execute the example code, pubsub_TSN_publisher.c. You will see the assert failure after pressing CTRL-C.

compile code:
gcc pubsub_TSN_publisher.c -I ../../src/pubsub -I ../../deps -lopen62541 -pthread -o pubsub_tsn

execute:
sudo ./pubsub_tsn lo

error output to console:
ua_architecture.h:165: UA_LOCK: Assertion `++(lock->mutexCounter) == 1' failed.
Aborted

Used CMake options:
 
```bash
cmake -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=300 -DUA_MULTITHREADING=199 \
      -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON \
      -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_IMMUTABLE_NODES=ON \
      ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): commit e5eba7b (current master HEAD as of Jan 13, 2021)
 - [x] Other OPC UA SDKs used (client or server): n/a
 - [x] Operating system: linux (Ubuntu)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: not relevant
 - [x] Wireshark network dump attached: not relevant
 - [x] Self-contained code example attached: provided example code
 - [x] Critical issue: high priority since it's causing our application to crash

jpfr, 2021-01-17:
Looking into this.
At this time, there is no multithreading support in the PubSub subsystem.

Let's see whether we have the capacity to add thread-safety for PubSub already for the 1.2 release.

rexsum, 2021-01-25:
The issue is that my application contains both client/server and pubsub capabilities and has been using the multi threading option for over a year. A change was made to the library not too long ago which is now causing this assert error to occur. I should not have to turn off multi threading capabilities because of this bug introduced into the pubsub subsystem.