open62541/open62541 #4122: About hiding the nodes of the address space in UaExpert

xujienan, 2021-01-14:
## Description
Hello everyone,
Recently, I encountered a problem. By modifying the allowBrowseNode_default function, I have achieved the permissions of different clients to browse nodes, but what I want to do is modify the client’s permission to see nodes. Therefore, I want to know whether the nodes in the address space in UaExpert can be hidden (ie invisible) ? What should I do in the server?

The picture shows the situation when different clients are browsing. The first client can browse "john" and "peter", and the second client can browse "mike". What I want to achieve is that only "Server", "john" and "peter" can be seen under the Root/Objects path of the first client, and only "Server" and "Server" can be seen under the Root/Objects path of the second client. "Mike".
Thanks.
[image]

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): UAExpert Client
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2021-01-15:
To hides nodes from the user, you need "Roles" Permissions. At the moment this isn't implemented in open62541. See: #3612

NoelGraf, 2021-04-16:
I would then close the issue. If something is still unclear, the issue can be reopened.

closed 2021-04-16 by NoelGraf