open62541/open62541 #4123: Assertion for repeated callbacks

basyskom-jvoe, 2021-01-14:
## Description
Servers built with open62541 1.2 terminate sporadically with the following message:
```/home/jvoe/open62541/src/server/ua_server.c:670: serverExecuteRepeatedCallback: Assertion `server->serviceMutexCounter == 0' failed.```

## Background Information / Reproduction Steps
Build the open62541 1.2 branch with the following CMake options
```bash
cmake -DUA_ENABLE_ENCRYPTION:BOOL="1" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_AMALGAMATION:BOOL="0" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL="1" -DUA_ENABLE_ENCRYPTION_OPENSSL:BOOL="1" -DUA_MULTITHREADING:STRING="100"
```
and link it to a server with one active repeated callback.
## Checklist

Please provide the following information:

 [X ] open62541 Version (release number or git tag): 1.2 (e401ddff84673dd49c8d2768fdda303e796b8fdb)

rexsum, 2021-01-14:
This issue looks similar to the one that I reported yesterday (issue: 4119). it looks like code was pushed that is causing these multi-threading assert errors to suddenly start showing up. I'd like for whomever it is that reviews these issues to be aware that this is a problem that needs to be addressed.

jpfr, 2021-01-17:
Please provide a backtrace from a debugger.
Should be easy to fix then.

basyskom-jvoe, 2021-01-22:
Unfortunately, I am unable to reproduce the issue with gdb attached.

o6-nbeier, 2026-10-20:
Thank you for the report.

The false-positive assertion was removed in #4267, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier